andi34/photobooth #1: Todo: fix mobile theme QR after taking picture.

andi34, 2019-06-05:
[image]

andi34, 2019-06-05:
QR from Gallery is fine afterhttps://github.com/andi34/photobooth/commit/c239718b93a86bcbb9c20f295cc534a46dd48c1b

[image]

andi34, 2019-06-05:
@F4bsi me Idee?

F4bsi, 2019-06-06:
@andi34 Besteht das Problem noch?
Müsste mich auch erstmal wieder reinfuchsen. Denke am Wochenende hätte ich wieder ein bisschen Zeit. 
Meistens ist aber in dem Fall das Problem, dass im CSS die Kombination aus `position: abolute;` und `left: 50%; margin-left: -225px; ` genutzt wird. Durch den festen Wert passt es nur auf Desktops / größeren Devices. Umgehen kann man das ein bisschen mit `transform: translate(-50%, -50%);` oder halt in den Media Querys

andi34, 2019-06-06:
Super, vielen Dank nochmal. Habe gestern stundenlang versucht es hinzubekommen, aber am Ende war es ein riesen weisse Fläche mit kleinem QR und dafür dann auch in der Galerie nicht mehr schön 🙈 raffe auch nicht warum der QR mit seiner Größe überhaupt sozusagen 2x definiert ist...

andi34, 2019-09-27:
Ready for review https://github.com/andreknieriem/photobooth/pull/78

closed 2019-09-27 by andi34