andi34/photobooth #4: Todo: add Download button

andi34, 2019-06-13:
Not many people i know use QR, user can connect  to photobooth via webpage and take pictures - let them also download them.

andi34, 2019-06-16:
Idea: if qr is disabled show generated download link which qr uses but don't show the QR picture?

thymon13, 2019-06-17:
Many people use QR at every wedding.
They love this simplicity there.

But my comment has nothing to do here I guess :)

andi34, 2019-06-17:
Don't worry, like said "not many I know" ;)

andi34, 2019-07-10:
Does not work, but keep here to play with it later again:
```
diff --git a/index.php b/index.php
index b29a375..2fdb559 100644
--- a/index.php
+++ b/index.php
@@ -185,6 +185,7 @@ require_once('db.php');
                                        <button class="pswp__button pswp__button--fs" title="Toggle fullscreen"></button>
                                        <button class="pswp__button pswp__button--zoom" title="Zoom in/out"></button>
                                        <?php if($config['use_mail']){ echo '<button class="gal-mail" title="Per Mail senden"><i class="fa fa-cloud-download"></i></button>'; } ?>
+                                       <button class="gal-download" title="Download"><i class="fa fa-download"></i></button>
                                        <?php if($config['use_print']){ echo '<button class="gal-print" title="Drucken"><i class="fa fa-print"></i></button>'; } ?>
                                        <?php if($config['use_qr']){ echo '<button class="gal-qr-code" title="Qr Code öffnen"><i class="fa fa-qrcode"></i></button>'; } ?>
                                        <!-- Preloader demo http://codepen.io/dimsemenov/pen/yyBWoR -->
diff --git a/resources/css/style.css b/resources/css/style.css
index 1b8f4eb..91ef491 100644
--- a/resources/css/style.css
+++ b/resources/css/style.css
@@ -758,7 +758,7 @@ hr.small {
 }
 
 /* line 645, ../sass/style.scss */
-.gal-qr-code, .gal-print, .gal-mail {
+.gal-qr-code, .gal-print, .gal-mail, .gal-download {
   color: #fff;
   background: transparent;
   font-size: 26px;
diff --git a/resources/js/core.js b/resources/js/core.js
index 00dc582..68c43d2 100644
--- a/resources/js/core.js
+++ b/resources/js/core.js
@@ -363,6 +363,17 @@ var photoBooth = (function () {
         })
     });
 
+    // download from in gallery
+    $(document).on('click touchstart', '.gal-download', function (e) {
+        e.preventDefault();
+        var img = pswp.currItem.src;
+        img = img.replace(imgFolder+'/', '');
+        $.ajax({
+            url: 'http://'.$_SERVER['HTTP_HOST'].'/download.php?image=' + encodeURI(img),
+        }).done(function (data) {
+        })
+    });
+
     // Send Mail gallery
     $('.gal-mail').on('click touchstart', function (e) {
         //e.preventDefault();
diff --git a/resources/sass/style.scss b/resources/sass/style.scss
index 47516cd..8766a1e 100644
--- a/resources/sass/style.scss
+++ b/resources/sass/style.scss
@@ -676,7 +676,7 @@ hr.small {
   }
 }
 
-.gal-qr-code,.gal-print, .gal-mail {
+.gal-qr-code,.gal-print, .gal-mail, .gal-download {^M
   color: #fff;
   background: transparent;
   font-size: 26px;
```

andi34, 2019-07-17:
Adjusted, need to test:
```
From a65b10ac46f9351edebf34ba08cacf06c9b3bb88 Mon Sep 17 00:00:00 2001
From: Andreas Blaesius <andi@***.org>
Date: Wed, 10 Jul 2019 17:13:11 +0200
Subject: [PATCH] add dl button to gallery

---
 index.php                 |  1 +
 resources/css/style.css   |  2 +-
 resources/js/core.js      | 11 +++++++++++
 resources/sass/style.scss |  2 +-
 4 files changed, 14 insertions(+), 2 deletions(-)

diff --git a/index.php b/index.php
index b29a375..2fdb559 100644
--- a/index.php
+++ b/index.php
@@ -185,6 +185,7 @@ require_once('db.php');
 					<button class="pswp__button pswp__button--fs" title="Toggle fullscreen"></button>
 					<button class="pswp__button pswp__button--zoom" title="Zoom in/out"></button>
 					<?php if($config['use_mail']){ echo '<button class="gal-mail" title="Per Mail senden"><i class="fa fa-cloud-download"></i></button>'; } ?>
+					<button class="gal-download" title="Download"><i class="fa fa-download"></i></button>
 					<?php if($config['use_print']){ echo '<button class="gal-print" title="Drucken"><i class="fa fa-print"></i></button>'; } ?>
 					<?php if($config['use_qr']){ echo '<button class="gal-qr-code" title="Qr Code öffnen"><i class="fa fa-qrcode"></i></button>'; } ?>
 					<!-- Preloader demo http://codepen.io/dimsemenov/pen/yyBWoR -->
diff --git a/resources/css/style.css b/resources/css/style.css
index 1b8f4eb..91ef491 100644
--- a/resources/css/style.css
+++ b/resources/css/style.css
@@ -758,7 +758,7 @@ hr.small {
 }
 
 /* line 645, ../sass/style.scss */
-.gal-qr-code, .gal-print, .gal-mail {
+.gal-qr-code, .gal-print, .gal-mail, .gal-download {
   color: #fff;
   background: transparent;
   font-size: 26px;
diff --git a/resources/js/core.js b/resources/js/core.js
index 00dc582..68c43d2 100644
--- a/resources/js/core.js
+++ b/resources/js/core.js
@@ -363,6 +363,17 @@ var photoBooth = (function () {
         })
     });
 
+    // download from in gallery
+    $(document).on('click touchstart', '.gal-download', function (e) {
+        e.preventDefault();
+        var img = pswp.currItem.src;
+        img = img.replace(imgFolder+'/', '');
+        $.ajax({
+            url: 'download.php?image=' + encodeURI(img),
+        }).done(function (data) {
+        })
+    });
+
     // Send Mail gallery
     $('.gal-mail').on('click touchstart', function (e) {
         //e.preventDefault();
diff --git a/resources/sass/style.scss b/resources/sass/style.scss
index 47516cd..8766a1e 100644
--- a/resources/sass/style.scss
+++ b/resources/sass/style.scss
@@ -676,7 +676,7 @@ hr.small {
   }
 }
 
-.gal-qr-code,.gal-print, .gal-mail {
+.gal-qr-code,.gal-print, .gal-mail, .gal-download {
   color: #fff;
   background: transparent;
   font-size: 26px;
-- 
2.7.4

```

ChrisKram, 2019-08-02:
I don't understand the use case of that button. As far as I know, the barcode shows the picture on your phone. What aren't you using the context menu to save the image? 

What shall a download button do and how will you work with that on a tablet or smartphone? For me as an iPhone Users I might not be able to save the files. 

for me a better use case is to upload the photos to a cloud share and share the download link via mail. So a form to add you email address might be a good idea so far. 

correct me if I misunderstand it.

andi34, 2019-08-02:
Different user , different use case.

Why should I do complicated using a QR code if I have access on the gallery and can click simply on "Download" to get the file on my phone?

It's a feature I need for myself/my use case.

closed 2019-12-10 by andi34