andi34/photobooth #5: Photobooth Title

tobiashaas, 2019-06-25:
Hey @andi34 ,
Vielen Dank dass du in letzter Zeit soviel für die Photobooth Software gemacht hast! 
Mich hat es sehr gefreut, da ich meine Box bald auf einer Hochzeit von meinem Kumpel einsetzen möchte ;-)

Meinst du es wäre möglich eine Art Titel einzubauen?
Also z.B 

Hochzeit von Max & Moritz am 12.12.12 ? 
Oder vielleicht sogar auch so etwas wie Filter wie man es von Instagram kennt, oder sogar eine Funktion um zb 3 oder 4 Fotos als Collage aufzunehmen? 

Das wäre sehr cool

Vielen Dank und ich freue mich von dir zu hören

andi34, 2019-06-25:
Hey!
Klar ist ein Titel möglich "einzubauen". Privat gerade viel um die Ohren und wenig Zeit für PC Sachen... Wenns passt schaue ich aber gerne nach.
Nächste Woche kommt mein Raspberry 4 hoffentlich und da möchte ich natürlich auch als bald als möglich mit spielen.

Mit Collage: mit ImageMagick vllt umsetzbar.
https://imagemagick.org/script/montage.php

Beste Grüße

Andi

tobiashaas, 2019-06-25:
Hi Andi, 
Super ich freue mich! 

Mein Rpi4 sollte morgen eintreffen <3

Das mit der Collage schaue ich mir mal an, danke 

Liebe Grüße und schönen Abend
Tobias

andi34, 2019-06-26:
Im lang/de.js

"startScreen": "<h1>Photobooth<\/h1><h2>Hochzeit von Max und Moritz und <\/h2>&copy; Die LIGA",

tobiashaas, 2019-06-26:
Schaue ich mir heute Abend mal an! Danke dir
Darf ich fragen warum du das überhaupt "mobil" anpasst?
Macht doch irgendwie keinen Sinn oder nicht? :-D 

Man benötigt ja eig nur den QR Code um an das Foto zu kommen, der rest wird ja an der PHotobooth selbst erledigt oder verstehe ich da was falsch :P

andi34, 2019-06-26:
Ging ums antworten hier von mobil. 😛

andi34, 2019-06-26:
Aber grundsätzlich: die Photobooth lässt sich mit meinem OnePlus 6 ziemlich gut bedienen. QR Code sieht nur nicht gut aus....

Anstelle von QR fände ich die Möglichkeit per Download button besser. Der Pi als AP eingerichtet reicht aus, könnte jeder der verbunden ist sein Bild runterladen. Kenne kaum wen, der QR nutzt....

tobiashaas, 2019-06-27:
Ja QR ist so ne Sache.... Aber ich hab schon Kopfschmerzen wenn ich nur daran denke, dass sich die Leute erstmal mit der Box verbinden müssen :-D Das checken viele auch wieder nicht. Und dann denken die noch sie haben Internet.... Weil WLAN ja für die meisten Internet bedeutet xD xD
Bin mal gespannt wie das wird, bald wird sie getestet!

tobiashaas, 2019-06-27:
> Im lang/de.js
> 
> ```
> "startScreen": "<h1>Photobooth<\/h1><h2>zur Hochzeit von Max und Moritz<\/h2> 12.12.12",
> ```
> 
> Reicht dir das?
> 
> Edit:
> Ich hasse mobile Seiten...

Ich hoffe ich komme heute dazu es zu testen! Cool wäre es wenn man das ganz Easy im Admin Panel ändern könnte :P ... Ich gebe dir Rückmeldung

andi34, 2019-06-27:
Das sollte kein Problem sein das im Adminpanel konfigurierbar zu machen, am Ende halt nur im Zweifel schwierig, da es dann in allen Sprachen so ist wie eingestellt

tobiashaas, 2019-06-29:
Hey Andi, jetzt habe ich das mal getestet und endlich gerafft wofür das genau ist :-D 
Klar, das ist natürlich fürs erste vollkommen okay ;-) 
Aber schön wäre ja wenn man das im Admin Menü ändern könnte - auch für alle Sprachen.

Damit kann ich aber erstmal arbeiten! Danke dir

andi34, 2019-06-29:
Kein Thema :)

andi34, 2019-06-29:
https://github.com/andi34/photobooth/commit/c8d329c382058f26a54d79e64c159a553b5f71af

Muss nur die config noch nen passenden Namen im Adminpanel bekommen.

Vorschläge?

andi34, 2019-06-29:
Erledigt :)

https://github.com/andi34/photobooth/commit/572b86eeb419ef016117cd7703b063f606f4a0e5

[image]

tobiashaas, 2019-06-29:
Gefällt mir sehr gut! Danke ;-)

andi34, 2019-06-30:
Zu den filtern: bin dabei mit zu spielen, leider zu resourcen-hungrig... Dauert ewig... Mal sehen ob ich da was auf die Beine bekomme

tobiashaas, 2019-06-30:
Super ich bin gespannt! Hab es mir auch mal angeschaut.... werde aber leider nicht schlau daraus ... bin leider absolut kein Coder ... Würde dir da gerne helfen ... bin aber ganz am Anfang :-(

andi34, 2019-06-30:
Graustufen Bild, Sepia und negativ könnte ok sein. Hatte aber größeres vor was leider, zumindest auf dem Pi3, nicht Umsetzbar ist (nicht gut ne Minute oder länger auf ein Bild zu warten)

andi34, 2019-06-30:
Das Ding ist:
Wenn mans ausdrucken will direkt ist Filter ok.
Wenn man's runterladen kann: Filter kann jedes Handy.
Und Filter kann man jeder Zeit im Nachhinein.

Klar, ne lustige Spielerei im Zweifel, aber muss auch praktikabel sein am Ende und der Pi ist eben kein High-Ende-Gerät.

Auf nem festen Linux PC mag das ganze ohne jegliche Probleme klappen.

Muss Mal sehen, was man wie am geschicktesten macht ... Bin mir da unschlüssig, auch wie viel Zeit ich investiere in die Option um sie anständig hinzubekommen

tobiashaas, 2019-06-30:
Da gebe ich dir vollkommen recht, dann lieber ohne Filter als ne Minute auf das Foto zu warten... 
Kannst du mir mal sagen was ich ins Terminal reinhacken kann um das zu testen? 
Dann sag ich dir mal wie lange der RPi4 dafür braucht ;-)

andi34, 2019-06-30:
```
sudo apt-get install imagemagick
convert meinjpg.jpg -bordercolor white -background black +polaroid

Nur als bsp

closed 2019-06-30 by andi34

tobiashaas, 2019-06-30:
Also wenn das Foto in der Auflösung nicht mega groß ist, läufts ganz flüssig.
In Voller Auflösung - was meine Canon 600d hergibt - hängt sich der RPi4 irgendwann auf... 
50% kleiner gehts super

andi34, 2019-07-01:
Hatte auch schon überlegt die Bilder einfach in der Größe vorher zu beschränken und dann umzuwandeln. Ist aber auch am Ende eher unbefriedigend...

```
convert myjpg.jpg -resize 50% myjpg.jpg

convert myjpg.jpg -bordercolor white -background black +polaroid myjpg.jpg
```
Das sollte dann auch am Ende immer das Original überschreiben.

andi34, 2019-07-01:
Glaub ich habe eine "bessere" Alternative für Standard Filter gefunden die ich die Tage testen werde.

Imagemagick hat generell auch Probleme aus PHP raus ausgeführt zu werden...

Ggf später die php Version nochmal mit installieren und schauen ob es damit geht. Den Polaroid Effekt finde ich nämlich ziemlich nice an sich.

tobiashaas, 2019-07-01:
Also diesen Polaroid Effekt finde ich auch ziemlich nice! Der hat's mir echt auch angetan.
<3

andi34, 2019-07-03:
Sepia, negativ und graustufen Bilder passt.
Muss nur den Code noch schick hinbekommen... und danach ins Admin panel vorerst. Vermutlich....

koenigd79, 2019-07-04:
Hi
Filter fände ich auch genial. 
Habe auf Andre Rina weibseite in den Kommentaren folgende Seite gefunden:
https://www.reinhard-rebernig.at/website/websites/fotokasterl/#
da finde ich ist das ganz toll umgesetzt mit den FIltern, der Autor Reinhard Rebernig würde auch teile seines Codes hergeben, ggf. erspart euch das ne menge Arbeit. Ich selber kann leider nicht wirklich programmieren und lese hier aber interessiert überall mit, da ich die Fotobox ne super Sache finde.
Hoffe es hilft....
Gruß Daniel

andi34, 2019-07-04:
Schon gesehen und in Kontakt getreten vor 2 Tagen. Muss Mal sehen was davon auf dem Pi läuft. Er lässt mir Code-Teile demnächst zukommen.
Imagemagick z.B. ist zu resourcen-hungrig... 
Da ich selbst noch Neuling mit PHP etc bin werde ich ggf. Hilfe brauchen, vor allem wenn der Code sauber seien soll (Code Qualität ist mir wichtig).

koenigd79, 2019-07-04:
Super,

ggf. hilft auch das noch weiter:

https://github.com/zaachi/PHP-Instagram-effects

damit habe ich vor einiger Zeit mal expiremtiert, aber kam nicht richtig zum Ergebnis mangels  php Kenntnis.

andi34, 2019-07-04:
Danke, habe schon was am laufen mit Standardfiltern, bin aber Codeseitig noch nicht ganz zufrieden und muss dazu auch noch alles ins Adminpanel packen bis ich eine bessere Lösung habe.

Kann also noch etwas dauern bis ich da was publiziere...

andi34, 2019-07-04:
Im übrigen hat André auch meine Pullrequest gemerged. Abgesehen von E-Mail ist also alles auch bei ihm jetzt im Source.

Email muss ich noch testen, noch nicht zu gekommen bisher 😆

andi34, 2019-07-20:
@tobiashaas habe nun die Filteroption soweit fertig. Werde ich demnächst dann hochladen für jedermann.

Kannst du zufällig https://github.com/andi34/photobooth/issues/11 bestätigen?

tobiashaas, 2019-07-20:
Werde ich heute mal testen! 😉

andi34, 2019-07-20:
@tobiashaas hab den Fehler gefunden. Habe oben den Commit gepostet.

tobiashaas, 2019-07-20:
Danke für die Info! Hab es eben gesehen ;-)