andi34/photobooth #7: HTTP ERROR 500 - Index.php

tobiashaas, 2019-06-29:
Hi Andi,

Wenn ich deinen Fork nehme, habe ich mit der "index.php" und auch der "ressources/js/core.js" Probleme die Seite zu öffnen.

Hab es mit dem Master getestet. Sobald eine der Dateien hochgeladen wird funktioniert die Seite nicht mehr.

Grüße ;-)

andi34, 2019-06-29:
Nimm Mal die private branch.

https://github.com/andi34/photobooth/commit/4496a9ca6ce1b807a4b7cdabed55ed19f08d37f0

Der ist neu und noch nicht getestet, wenns nicht geht einfach einmal reverten.

Vorher nur die config.js im Admin Ordner löschen.

andi34, 2019-06-29:
Schaue mir das ganze nachher an. Mit Glück kommt mein Pi4 heute auch schon. 😍

andi34, 2019-06-29:
Okay, habe die fehlerhaften 2 commits rausgehauen. Extra branch bis ich's gefixt habe.
Private branch geht somit wieder ;)

closed 2019-06-29 by andi34

andi34, 2019-06-29:
@tobiashaas hab noch etwas am Code angepasst grundsätzlich. Für heute bin ich fertig. Meine "private" branch funktioniert wunderbar mit allen Neuerungen.

Rest die Tage wenn Zeit ist :)

tobiashaas, 2019-06-29:
Super ich danke dir vielmals! Finde ich super dass du das Projekt voran bringst! 
Werde es morgen auf meinem neuen RPi4 testen <3

andi34, 2019-06-29:
Kein Thema. Lass mich Mal wissen ob's läuft auf dem 4er mit Raspbian Buster. 😊

Mein 4er hängt leider übers Wochenende noch bei DHL ab...

tobiashaas, 2019-06-29:
Also soweit läuft alles, bis auf das Foto machen ... 
Bekomme beim Foto schießen eine Fehlermeldung... evtl. stimmt was mit den Rechten nicht.

Für die Installation ist eine kleine Änderung notwendig - oder auch nicht - da das Paket schon integriert ist.
NEU`sudo apt-get install git apache2 php php-gd ffmpeg`
ALT`sudo apt-get install git apache2 php php-gd libav-tools`

andi34, 2019-06-30:
Danke :)

Hab die README entsprechen angepasst.
Habe heute morgen die Leere Gallerie Nachricht aus dem Admin panel entfernt und in den lang Files gepackt. Denk nicht das die jemand anpassen will (eh nur bis zum 1. Foto zu sehen) und wenn doch eben entsprechend der Sprache ;)

tobiashaas, 2019-06-30:
Hey Frage,
Hast du was mit dem gPhoto2 geändert?
Hab heute das Problem gefunden warum er auf dem RPi4 zwar ausgelöst, aber die Fotos nicht gespeichert hat...

Hatte keine SD Karte in der Kamera :-D Oder war das jetzt nur Zufall?
Falls nein könnte man das evtl in der Readme noch hinterlegen. 
Bei Andre im Forum gibts nämlich viele solcher Anfragen! :-)

Schönen Sonntag

andi34, 2019-06-30:
https://github.com/andi34/photobooth/commit/62e4e12ef25c2a300a871a5c3a389c55af3da761

https://github.com/andi34/photobooth/commit/d5388faab13c34bc43733dae5cccc637d21c9013

Die private branch ist mein persönliches Setup und Default für mich. Werde ggf. Die dev2 branch default machen wenn André meine Pullrequest nicht merged und dann selbst weiter maintainen.

Instruktionen sind dafür aber auch im README schon drin: https://github.com/andi34/photobooth/commit/c3f7eff3e27a333ff6883fe541c27245e365fd6f

Ich habe die Bilder lieber immer an 2 Orten ;)

Dir auch einen schönen Sonntag!

tobiashaas, 2019-06-30:
> Ich habe die Bilder lieber immer an 2 Orten ;)

Da bin ich ganz bei dir! Sicher ist sicher :-D

commit c4ace13c (2019-11-18): Add download picture button to gallery

fix #7