andi34/photobooth #11: Bug: gallery settings not saved

andi34, 2019-07-16:
All other settings save fine. Need to investigate further, no quick fix found.

andi34, 2019-07-20:
Fixed https://github.com/andi34/photobooth/commit/0a8c874e5ad5fa93f9c1eb61b9b8746954d2f313

closed 2019-07-20 by andi34

tobiashaas, 2019-07-20:
Also bei mir wird es nicht gespeichert ... Hab es jetzt mehrfach heruntergeladen und neu draufgepackt! Dev und Master...
Email, Galerie und Hochzeit speichert er nicht

andi34, 2019-07-20:
Komisch. Bei mir läuft es auf Buster und auf Xubuntu LTS.

Hast du die config.js vorher gelöscht?

andi34, 2019-07-20:
Ggf auch einmal neustarten den Pi.

andi34, 2019-07-20:
Im Zweifel auch:
```
sudo usermod -a -G www-data Pi
```

tobiashaas, 2019-07-20:
> Komisch. Bei mir läuft es auf Buster und auf Xubuntu LTS.
> 
> Hast du die config.js vorher gelöscht?

Die Config wars ..... die hatte ich vergessen zu löschen! Jetzt passt es
Danke dir

andi34, 2019-07-20:
Sehr gut :) freut mich :)

Nur noch Kleinigkeiten, dann gibt's auch die Filter. 🥳

Wann ist die Hochzeit wo du die Box nutzen willst?

tobiashaas, 2019-07-20:
Die Hochzeit war bereits am 13.7 ;-) 
Hab dieses Jahr aber noch 2 Hochzeiten von Freunden wo ich die Box gerne aufstellen würde! 

Hat sich bei knapp 200 Fotos mal aufgehängt.
Ich suche aktuell noch nach einer Möglichkeit, bei einem Touchscreen in Chromium, die Zoom Funktion zu deaktivieren. Die Box lies sich nicht mehr bedienen nachdem jemand reingezoomt hat... ;-)

tobiashaas, 2019-07-20:
Auf die Filter bin ich gespannt! <3

andi34, 2019-07-20:
Mobile Seite? Oder über die Desktop Seite?
Da kein responsive Design erzwingt es in meinem tree die desktop Seite nun.

andi34, 2019-07-20:
```
<meta content="width=device-width, initial-scale=1.0, maximum-scale=1.0, user-scalable=no" name="viewport"/>
    <meta name='apple-mobile-web-app-capable' content='yes' />
```

tobiashaas, 2019-07-20:
Auf der Desktop Seite

tobiashaas, 2019-07-20:
[image]

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

tobiashaas, 2019-07-20:
Auf der Mobilen Seite - auf meinem iPhone XS - ist die Galerie nicht nutzbar.

andi34, 2019-07-20:
Die Ansicht liegt daran, das die Booth nicht für mobile Geräte geschrieben ist. Auf der Desktop Seite sollte alles aussehen wies soll.

Responsive Design ist ne Sache.... Aber sehr aufwendig und alleine schaffe ich das nicht. Dafür bin ich nicht weit genug mit meinen Kenntnissen.
Teile sind ja auch im core.js hinterlegt und garnicht nur über die style.css definiert. Macht es nicht einfacher.

andi34, 2019-07-20:
Fork deaktivert fix den Zugriff auf den button?

tobiashaas, 2019-07-20:
Mir reicht die Desktop Seite, wollte dich nur darauf hinweisen ;-)
Ne der Fork ist nicht der Übeltäter

tobiashaas, 2019-07-20:
Ich muss mich auch mal bisschen mehr in die Thematik reinfuchsen.
Möchte auch mehr helfen können! ;-)

andi34, 2019-07-20:
Responsive Design ist auf meiner Liste ;)

Bis dahin:
Kannst natürlich die Schrift in der index.php anpassen. Aus <H1> z.B. <h2> machen oder gar <p>

Dann sollte auch mehr Schrift in eine Zeile passen ;)

andi34, 2019-07-20:
Ist leider auch so ziemlich alles sehr abhängig von der Display Größe...

andi34, 2019-07-20:
Versuche immer alles in logischen commits hochzuladen. 
So lerne ich immer: Anschauen was wird wie umgesetzt. Ist es ein logischer commit der nur eine Sache macht ist vieles einfach nachvollziehbar. Würde ich alles in einen commit packen und "Update" nennen hast du tausend teile die du verstehen musst.

Mache das seit Jahren so im Android Bereich auch und bin damit schon weit gekommen.

tobiashaas, 2019-07-20:
Wäre es eigentlich nicht sinnvoller, wenn man das einfach an ein fertiges Framework anpasst, welches komplett Reponsive ist? 
Hab heute mal ein bisschen gegoogelt und bin auf solche Frameworks gestoßen 😉
https://www.keycdn.com/blog/front-end-frameworks

Liebe Grüße 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

andi34, 2019-07-20:
Kein plan, wie viel Mehrarbeit oder Wenigerarbeit das ist 😅

tobiashaas, 2019-07-20:
Wenn ich die Tage mal Zeit habe schaue ich mir das mal genauer an ;-) dass ich auch mal wieder was dazu beitragen kann 🙈

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

andi34, 2019-07-20:
Sollte das meiste mit den Media querys im style.css gehen. Im Browser lässt sich ja auch eine Gerätegröße einstellen und somit am PC anpassen möglich sein .