andi34/photobooth #12: Add Imagefilter

andi34, 2019-08-01:
[image]

[image]

andi34, 2019-08-01:
@tobiashaas @koenigd79

andi34, 2019-08-02:
Pushed on all branches

closed 2019-08-02 by andi34

koenigd79, 2019-08-06:
Hi,

so jetzt mal Filter probiert habe, klasse, funzt.
Was mich allerdings wundert habe jetzt auf nen RPi4 4gb gewechselt und auf deine Version und zw. Foto und der Anzeige vergehen fast 10 Sekunden, vorher waren es 5 sec. Bei Verwendung der Filter nochmal 2-3 Sekunden mehr.
Kannst du das bestätigen??

andi34, 2019-08-06:
Hab die Zeit nicht gestoppt, aber hatte auch das Gefühl daß es auf dem Pi4 etwas Träger ist im Vergleich mit dem Pi3.

Haste Mal geschaut obs mit Andrés branch schneller funzt?

koenigd79, 2019-08-07:
Hi, so jetzt mal Andres branch probiert, gleiches Porblem, braucht auch 10 Sekunden bis zur Anzeige des Bildes..... egal ob auf direkt angeschlossenem Display oder auf Tablet. Komisch. Mehr Leistung + 4GB Ram da habe ichmehr erwartet.... werde mal schauen wo es dran hängt. Wenn ich was finde gebe ich Rückmledung. 
Gruß Daniel

andi34, 2019-08-07:
Ggf am neuen Raspbian?
Oder es ist eben noch nicht alles optimal im Quellcode (Kernel?). Pi3 erfährt ja schon ne Weile Aufmerksamkeit.

devnull1975, 2019-08-07:
Ich habe für die Fotobox noch einen "alten" Pi3 laufen. Zu Anfang hatte ich eine alte 16GB SD-Karte drin mit der es bei ca. 3 MB Fotos zwischen 5 und 6 Sekunden dauerte bis das Foto auf dem Tablet via WLan (Pi als AP) angezeigt wurde. 
Mit einer schnelleren 32GB Karte hab ich nun die Auflösung der Fotos auf knapp 8 MB stehen und es dauert genau so lange bis das Foto angezeigt wird.
Als OS lauft bei mir Buster.

Schon komisch das der neue Pi länger braucht. Irgendwo hab ich gelesen das der Pi4 bei hoher Leistung sehr war wird und eigentlich einen Lüfter bräuchte. Eine neue Firmware soll das Hitzeproblem wohl etwas beheben. Vielleicht könnte das ja helfen, sofern nicht schon probiert.

firesta666, 2020-08-06:
Hallo alle zusammen,

ich nutze die Photobooth schon seit über 1 Jahr und bin immernoch sehr begeistert. 

Nun habe ich auf meinem PI4 8GB das Problem, das mit auf der Hauptseite die Bildfilter an der rechten Seite nicht eingeblendet werden, sondern nur nach dem erstellten Bild. Im Adminbereich habe ich es aktiviert: Bildfilter erlauben.
Woran kann das liegen? Beim meiner alten Version und so auf dem RPI3 wie es hier geschrieben wurde, soll es ja auch auf der Eingangsseite schon auswählbar sein. Kann mir jemand kurzfristig helfen? Habe morgen eine Veranstaltung und würde es sehr gerne einschalten. Danke euch.

andi34, 2020-08-06:
@firesta666 vorher den Filter auszuwählen ist nicht mehr möglich (nur einen definierten Filter immer anzuwenden), nur noch anschließend - dann sieht man auch direkt die Vorschau bevor das Bild abgespeichert wird in der Datenbank. Hat den Vorteil, dass du explizit am gemachten Bild einen geeigneten Filter aussuchen kannst und nicht ins blinde einen auswählst.