andi34/photobooth #13: greenscreen funktioniert nicht

Nie-Oh, 2019-08-02:


andi34, 2019-08-02:
https://github.com/andi34/photobooth/commit/dc3c045370411d5a75756f2f9e2a0a06d97a6fdc

Bringt aber leider auch nichts, mit Hintergrund wird kein Bild gespeichert, nur ohne.

Bezüglich des Pinsel:
Ggf Mal den Browser Cache leeren :)

Nie-Oh, 2019-08-02:
Hab ich schon gemacht Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Betreff: Re: [andi34/photobooth] greenscreen funktioniert nicht (#13)Von: Andreas Blaesius An: andi34/photobooth Cc: Nie-Oh ,Author dc3c045
Bringt aber leider auch nichts, mit Hintergrund wird kein Bild gespeichert, nur ohne.
Bezüglich des Pinsel:
Ggf Mal den Browser Cache leeren :)

andi34, 2019-08-02:
Funktioniert hier ohne Probleme der Pinsel Button in der Galerie.
Admin Panel benutzt und config.js stört?

Muss wenns dann funktioniert noch die config ins Admin Panel packen

andi34, 2019-08-02:
Gebe für heute auf. Werde die Tage nochmal schauen wenn nicht jemand anderes schneller war/ist

Nie-Oh, 2019-08-02:
der Pfad wird falsch zusammen gebaut

http://localhost/chroma_keying/index.php?filename=%5Cimages%5C20190802_131826.jpg&location=aHR0cDovL2xvY2FsaG9zdC8=

Muss Sein 

http://localhost/chroma_keying/index.php?filename=%2Fimages%2F20190802_131826.jpg&location=aHR0cDovL2xvY2FsaG9zdC8=

Also \ Entspricht %5C
Muss aber / also %2F sein

andi34, 2019-08-02:
Eigenartig... Finde es auf Anhieb nicht, woran es liegen könnte. Ich schaue die Tage nochmal rein :) danke dir schon Mal!

Nie-Oh, 2019-08-02:
Die url wird irgendwie falsch zusamnen gebaut. Die variable filename hst auch einen pfad mit drinVon meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Betreff: Re: [andi34/photobooth] greenscreen funktioniert nicht (#13)Von: Andreas Blaesius An: andi34/photobooth Cc: Nie-Oh ,Author Eigenartig... Finde es auf Anhieb nicht, woran es liegen könnte. Ich schaue die Tage nochmal rein :) danke dir schon Mal!

andi34, 2019-08-02:
https://github.com/andi34/photobooth/commit/0b9db0bf5d7a96a88f0cc6ee25e41ff3d3e396f4

Keine Ahnung ob's das fixt , kann ich erst morgen oder nächste Woche testen

andi34, 2019-08-04:
https://github.com/andi34/photobooth/commit/b799079f20d5b8c31c3e73a099c8321090ffec4b

andi34, 2019-08-06:
Bildgröße schien ein Problem zu seien. Geht nun alles anscheinend.

Closed. Wenn wieder ein Fehler auftritt gerne wieder öffnen.

closed 2019-08-06 by andi34