andi34/photobooth #17: White page / small errors

firesta666, 2019-08-08:
Hello Andi34, first thank you that you work actually on this project. I love the functions from the basic version from Andre Rinas and found now your version very cool and good new features of the version from yesterday.

My system:
RPI 4, 2GB RAM, 32GB MicroSD
Basic version 1.53 are running fine, but i found the new filters a cool function and update my system with your newest version. I have do it all more as two times before i investigate what are the issue.
After install your version, i can immediatelly look and change things in the /admin page, but the normal index.php are empty. Clear cache, restarts, reinstall, remove and install are not helping.
Open the browser and pop up in the in inspection area:
SyntaxError: expected expression, got ';'10.0.96.32:35:19 – / 10.0.96.32 is my IP address, 19 are the line, i think.
And this one:
ReferenceError: cntdwn_time is not defined
photoBooth http://10.0.96.32/resources/js/core.js:8
http://10.0.96.32/resources/js/core.js:904

If you need more information or something, i can provide it to you asap, lower 1 hour after your reply. Can you help here?

I have found a workaround:

In the index.php, i remove the line: 48 - var ir_trigger = ;

then the index.php starts with another error:
jQuery.Deferred exception: useIRTrigger is not defined @http://10.0.96.32/resources/js/core.js:144:13
e@http://10.0.96.32/resources/js/jquery-3.4.1.min.js:2:29453
l/</t<@http://10.0.96.32/resources/js/jquery-3.4.1.min.js:2:29755
undefined.

Another issue: If i disable in the /admin page the filters, i can´t take a picture.
Error:
TypeError: document.getElementById(...) is nullcore.js:97:11
closeNav http://10.0.96.32/resources/js/core.js:97
http://10.0.96.32/resources/js/core.js:300

With enabled filters, it works.

I am not a programmer of php but i can read it.

andi34, 2019-08-08:
Which branch you use?
Cloned from my git?

firesta666, 2019-08-08:
Your branch right. Use this to install them: 

`cd /var/www/
sudo rm -r html/
sudo git clone https://github.com/andi34/photobooth html
cd /var/www/html/resources/lib
sudo git clone https://github.com/PHPMailer/PHPMailer
cd /var/www/html
sudo cp config.inc.php my.config.inc.php
sudo mkdir -p /var/www/html/images
sudo mkdir -p /var/www/html/keying
sudo mkdir -p /var/www/html/print
sudo mkdir -p /var/www/html/qrcodes
sudo mkdir -p /var/www/html/thumbs
sudo mkdir -p /var/www/html/tmp
sudo chown -R pi: /var/www/
sudo chmod -R 777 /var/www`

andi34, 2019-08-08:
Thanks, I'll check in the evening

andi34, 2019-08-08:
Should work now. Quickly fixed. Copy paste Derp, sorry.

firesta666, 2019-08-08:
Thanks for your quick help. 

Have you see the other small problem: 

Another issue: If i disable in the /admin page the filters, i can´t take a picture.
Error: 
[image]

And i have one question, is it possible to do a serie of 2 or 3 pictures in a row?
2 pictures are first enough, that helps the peoples that they must run every time to the touch display and click to do it. 

i have tested all with your fixed version with "sudo git clone https://github.com/andi34/photobooth html"

firesta666, 2019-08-08:
another issue after use the new fixed version:

take a picture and this error message pops up in the developement console from the browser.
the time to create one picture is now around 6 sec., before the fixed version it was 3 seconds. 
I think the issue is here, that they search the public.stream and that hold the time, 

[image]

Many thanks to you and best wished from Germany :-)

andi34, 2019-08-08:
Sorry, missed that. Just at work, looking at it at home.

andi34, 2019-08-08:
About the pictures in a row, sure, can be added. But needs some time to implement it.

firesta666, 2019-08-08:
Thank you :-). My idea is, to add a button between the take a picture on the homescreen, then under that take x <-that is the value they can modify in the config or in the admin panel, pictures, then image filter.  

Two easy questions for you: 
Can you tell me, in which file i can change the time after the picture is created and displayed, back to the home screen? I would like to change it to a lower value, to solve more picture and the waiting of them. 
Can you tell me, in which file i can disable the image resizing?

andi34, 2019-08-08:
Pictures only resized on chroma key (Greenscreen).

It always creates a thumbnail, see takePic.php. better performance in Gallery Overview to use a thumbnail and only show the full resolution image if you click on the thumbnail.

About the time:
Must be resources/js/core.js

andi34, 2019-08-08:
https://github.com/andi34/photobooth/commit/e99d133f06e6667f78a0c459855aea152e82fe41

Should fix takePic if Imagefilter disabled

firesta666, 2019-08-08:
Hey Andi34, nice and very good work :-) All is working.

Thank you very much :+1:

closed 2019-08-09 by andi34

andi34, 2019-08-09:
@firesta666 about more pictures in a row you could check how it's handled here https://github.com/andi34/photo-booth/blob/master/app/booth.js#L112

firesta666, 2019-08-11:
Hello andi34, 

can you help me to implement it in your Photobooth. 

I can read php but i can´t programming it. A little help was very nice. 

Currently i have not installed a node on my RPI4 i am not sure if your version is running fine with them?

andi34, 2019-08-11:
@firesta666 thanks for your support :)

I can check again when i find time for it. Shouldn't be to complicated. I am also still learning php and js here 🙈 learning by doing. Working as a medical assistant in real life. And real life atm gets big changes and not to many time on my projects.