andi34/photobooth #20: Print confirmation.

thymon13, 2019-08-12:
Hi,

Is it possible to confirm with pop up or other when I click on print button ?
Currently nothing happens on the screen when you click on it. And people by reflex click again several times over.

Sorry for my English :)

Thanks in advance.

thymon

andi34, 2019-08-12:
https://github.com/andreknieriem/photobooth/issues/44

flighter18, 2019-08-12:
I implemented the print confirmation with sweetalert2.all.min.js
https://sweetalert2.github.io/#examples

1. download sweetalert2.all.min.js and save it to folder "resources"
2. add sweetalert2 to head of index.html
`       <script type="text/javascript" href='resources/sweetalert2.all.min.js'> </script>
        <script type="text/javascript">
                function printConfirmation() {
                        swal({
                                type: 'success',
                                title: 'Druckauftrag gestartet',
                                showConfirmButton: false,
                                timer: 1600,
                                allowOutsideClick: false
                        })
                }
`
3. Add "onclick" Function to the printbutton in index.html
`onclick="printConfirmation()"`

4. Change the gallery print button in index.html
`'<?php if($config['use_print']){ echo '<a style="padding-top:14px"  href="#" onclick="printConfirmation();" class="gal-print" title="Close (Esc)"><i class="fa fa-print"></i> </a>'; }?>`

5. Change the z-index from swal2-container in resoureces/css/style.css
.swal2-container {
  z-index: 20000;
  -webkit-overflow-scrolling: auto
}

now the only problem i have is, if i push the print button on a mobile device in the gallery,  
on one hand, the confirmation appears but on the other hand the webpage zooms in.
My thinking is, the push on the print button simulates a double tap -> so the mobile device zooms in. 
Maybe the best solution is, to close the picture preview after the bush, on the print button. 
Has anybody a clue how to close a picture preview in the gallery? I couldn't get it.

My problem: 
[https://youtu.be/LfzfPWZhYK4]

andi34, 2019-08-13:
I am quite busy this week, I'll take a look once I find time. I think there's an easier way, adding something like
```
$('.loading').text(L10N.print);
```
Inside core.js at the printing function

flighter18, 2019-08-13:
Thanks for your input.
I added my swal2 function inside core.js at the printing function and removed all onclick from the index.php. I also changed the gallrery printbutton to his defaults. 
It works perfect without any problem.

andi34, 2019-08-13:
Could you paste the output of:
```
git diff resources/js/core.js
```

flighter18, 2019-08-13:
this is core.js

```
+++ b/resources/js/core.js
@@ -164,7 +164,15 @@ var photoBooth = (function () {
             $.ajax({
                 url: 'print.php?filename=' + encodeURI(result.img),
             }).done(function (data) {
-            })
+                            swal({
+                        customClass: 'myPrintConfirmation',
+                        type: 'success',
+                        title: 'Druckauftrag gestartet',
+                        showConfirmButton: false,
+                        timer: 1600,
+                        allowOutsideClick: false
+                })
+           });
         });

         // Add Image to gallery and slider
@@ -371,7 +379,15 @@ var photoBooth = (function () {
         $.ajax({
             url: 'print.php?filename=' + encodeURI(img),
         }).done(function (data) {
-        })
+                        swal({
+                        customClass: 'myPrintConfirmation',
+                        type: 'success',
+                        title: 'Druckauftrag gestartet',
+                        showConfirmButton: false,
+                        timer: 1600,
+                        allowOutsideClick: false
+                })
+       });
     });

     // chroma keying print
```

flighter18, 2019-08-13:
had to add the z-index to resources/css/style.css for the gallery confirmation

@@ -844,3 +844,10 @@ hr.small {
     bottom: auto;
   }
 }
+.swal2-container {
+  z-index: 20000;
+  -webkit-overflow-scrolling: auto
+}
+.myPrintConfirmation .swal2-icon {
+ border-color: orange;
+}

flighter18, 2019-08-13:
finally i added the src to index.php

```
index 94fb420..43ade29
--- a/index.php
+++ b/index.php
@@ -10,6 +10,9 @@ require_once('folders.php');
 require_once('db.php');

 ?>
+<script src="resources/sweetalert2.all.min.js">
+}
+</script>
 <!DOCTYPE html>
 <html>
 <head>
```

andi34, 2019-08-13:
Thanks. I'll check if my solution also works as it would need less dependencies and would make it easier to maintain.
If not I'll add your solution until I find a better one :)

andi34, 2019-08-13:
@flighter18 i've added something to test on "dev" branch. I've no printer, could you please check if it works?

flighter18, 2019-08-13:
I didn't get any confirmation.

git status:
` Author: Andreas Blaesius <andi@unlegacy-android.org>
Date:   Tue Aug 13 19:51:35 2019 +0200
admin: fix print message and command `

andi34, 2019-08-13:
dev branch? Or master branch?

Mh... I'll reinvestigate

flighter18, 2019-08-13:
on dev branch

andi34, 2019-08-13:
Could you check again please, pushed another change

flighter18, 2019-08-13:
still no message.

andi34, 2019-08-13:
Thanks. I am only mobile now, will check on PC again and let you know if there's something more

flighter18, 2019-08-13:
perfect

andi34, 2019-08-14:
Not sure if it works now. Still only mobile and not near a PC the whole day.

andi34, 2019-08-14:
https://github.com/andi34/photobooth/commit/443b15eecf6c35d422896c975ccead78f823b9cb

Maybe have to adjust the class & id and also add some changes to style.css for changed class & id and id

thymon13, 2019-08-14:
@andi34 @flighter18 thank you a lot for your interest !!

flighter18, 2019-08-14:
@andi34 still no confirmation

andi34, 2019-08-14:
@flighter18 danke fürs testen. Oh man... Muss wohl doch die Tage am PC dann schauen.

@thymon13 it is currently WIP and does not work how I've tried to implement it. Have to check next days when I am on a desktop PC (currently mobile only). I don't speak french, but from my Android development i know some people from different countries who can translate for me into french. Some basic translation via Google translate.

andi34, 2019-08-14:
If I haven't forgotten anything this branch should work: https://github.com/andi34/photobooth/commits/printconfirm
Please note: you must have cloned from my GitHub:
```
git fetch origin
git checkout origin/printconfirm
```
I'd like to have a way implement in the end without depending on other projects to make maintaining easier in the end.

andi34, 2019-08-19:
Removed WIP stuff out of dev branch as dev branch is used by default on my GitHub.
I'll work on it when I have more time again.

thymon13, 2019-08-31:
hi,
I don't understood how I can test this branch ? I'm noob in GIT

andi34, 2019-09-01:
If you've cloned from my git you can use the instructions 2 posts above yours.
```
cd /car/www/html
git fetch origin
git checkout origin/printconfirm
```

andi34, 2019-09-01:
But stupid question:
Isn't it added already via
```
$config['print']['msg']
```
???

thymon13, 2019-09-02:
> If you've cloned from my git you can use the instructions 2 posts above yours.
> 
> ```
> cd /var/www/html
> git fetch origin
> git checkout origin/printconfirm
> ```

Ok thx. I had to confirm the commit
But nothing happen when I click "print" like before...

> But stupid question:
Isn't it added already via
> 
> ```
> $config['print']['msg']
> ```
>???

Oh yes? But nothing... :)

thymon13, 2019-09-03:
It works for you?

thymon13, 2019-09-06:
I am sorry to up the topic. 
But it does not work at all. Nothing happens when I click on print. Did I miss something?

I am on the branch "print confirmation"

andi34, 2019-09-06:
No printer to test here.
I'll update in Sunday, maybe you can test again

thymon13, 2019-09-06:
@andi34 no printer needed for test it

andi34, 2019-09-06:
https://github.com/andi34/photobooth/commit/a32e41ef1b9d6700353860f3ebe037c76a4f67b1

try again with that (branch updated)

thymon13, 2019-09-06:
nothing :( like before.

thymon13, 2019-09-07:
After another try: the print function does not work anymore with this branch

andi34, 2019-09-08:
After latest Change?
```
cd var/www/html
git revert e42ee38f1913b341cecc84daa5a8c4a6bc93cd2e
```
@flighter18 what am I missing here https://github.com/andi34/photobooth/commit/fdde9a0acac197b472b48f49ac635d7cf76428d7 ?

flighter18, 2019-09-08:
@andi34
I am on holiday right now, will be back next week 😊
So can‘t check it. Sorry

andi34, 2019-09-08:
No Problem, guess we can make the frame work from edge to edge.
Have nice holidays! I'll have some soon too finally 😅

flighter18, 2019-09-11:
had some time :) 
did a new git clone from the new version and added the function again. Should now work.

also did a new merge request

thymon13, 2019-09-11:
@flighter18 
I tested. Works well.
But on touch pad the display has a problem now
[image]

thymon13, 2019-09-12:
@flighter18 

Thank you !
Works when I take a picture.  But do not work from the gallery.

thymon13, 2019-09-13:
@flighter18 

After other tests, I see that this generates bug page layout on the home page.

flighter18, 2019-09-13:
@thymon13 I have no problem on my test system. What for a browser do you use? 

And is the color of the hook orange at the end (see pic below) or does it stay green? 
[image]

Could you send a print screen from the bug page layout?

thymon13, 2019-09-14:
@flighter18 
It does stay green, and it is executed twice.

https://youtu.be/qIaNuETD6eI

flighter18, 2019-09-15:
@thymon13 

thank you for the video. 
Are those lines in your style.css file?
```
.swal2-container {
  z-index: 20000;
  -webkit-overflow-scrolling: auto
}
.myPrintConfirmation .swal2-icon {
 border-color: orange;
}
```

the first one makes the print cofirmation visible in the gallery (z-index). The second one changes the color.

flighter18, 2019-09-15:
@thymon13 
btw.

After yesterday's use of the photobooth at a wedding, I will change the confirmation a little bit. 

My idea:
After pushing the print button the page will automatically leave "picture view" mode and return to the home page or gallery page + it will display the print confirm.

flighter18, 2019-09-15:
@thymon13 

Check out my repo. I added this close function:

https://youtu.be/NTxXxTOfN8U

could you test it?

thymon13, 2019-09-16:
@flighter18 
Private video...

flighter18, 2019-09-16:
@thymon13

😝🙈
Please check it again

thymon13, 2019-09-16:
@flighter18 
cool !
Yes I could test it...

flighter18, 2019-09-16:
@thymon13 
i did a nother push.

now the print confirmation message comes from the language file.

flighter18, 2019-09-16:
@thymon13 
be sure to remove your browser cache befor testing.
I had some trubels with my google chrome browser.

thymon13, 2019-09-16:
> @thymon13
> now the print confirmation message comes from the language file.

@flighter18 
Clean !

andi34, 2019-09-16:
@flighter18 could we try https://gist.github.com/andi34/48a5c4f5e21e02bd6921bf86ca8e1902 instead swal?
Less dependencies = less maintain time which makes it easier to update in future.
I'd try to adjust it with some of your changes later or tomorrow evening.

andi34, 2019-09-16:
Pushed to my dev branch. Not tested and needs adjustment.
Will be one commit later, quickly done at work 🙈

andi34, 2019-09-16:
Looks fine to me and like a clean solution.
https://github.com/andi34/photobooth/commit/3272cb7b96948d5aa9c6f6c353f392ddeebd87a2
https://github.com/andi34/photobooth/commit/bc3daf1c571124bf483b9b84dbb13b6a7a2a5de9

@flighter18 and @thymon13 could both of you please test?

thymon13, 2019-09-17:
@andi34 
Ok I'll keep you informed

andi34, 2019-09-18:
Found some nitpicks, will edit the commit one again in the evening

flighter18, 2019-09-18:
@andi34 
I also tested it. works like a charm :)
The nitpicks are just small things, so i think we have the sollution now!

andi34, 2019-09-18:
Thanks for testing. Going to merge it into Andre's Source after work 
:)

[1 more reply]

closed 2019-09-18 by andi34