andi34/photobooth #21: Falsche Variable wird überprüft

flighter18, 2019-08-18:
Hi @andi34 

Ich habe noch ein Buggentdeckt. Und zwar im takePic.php File in der Zeile 24.
liste du dort die Variable imgFilter aus. Jedoch wird sie im core.js als "filter" verschickt. Da die Variable dann als undefiniert angezeigt wird, werden die Fotos immer zuerst im tmp Verzeichnis gespeichert. 

Nach folglich noch die zwei Zeilen:

Zeile 24 im takePic.php
`if(!empty($_POST) && !($_POST['**imgFilter**'] == 'imgPlain')) {`

 Zeile 121 im core.js
` jQuery.post("takePic.php", { **filter**: imgFilter }).done(function( result ){`

andi34, 2019-08-19:
Danke :)

andi34, 2019-08-19:
https://github.com/andi34/photobooth/commit/acfedb0400f28c615fda6f9b7e9f2092ef7e20af

closed 2019-08-19 by andi34