andi34/photobooth #22: Todo: Boarder around the picture

andi34, 2019-08-22:
Adjust and optimize source from Steffen Musch.

Need to know: Example boarder can be used without trouble? Source & license?

Nie-Oh, 2019-08-22:
@andi34 

it is better you ceate a new Example Boarder. 

Steffen

andi34, 2019-09-08:
frame.png needs to be added to root of photobooth.
https://github.com/andi34/photobooth/commit/20a9b16623431942aed1610e843ae6f2e665366b#diff-d5fde551bb3c77d93baa17a8ba21e781R58

If someone can share a frame which can be used in this project without license issues let me know.
No frame.png added by default until that.

andi34, 2019-09-08:
[image]

andi34, 2019-09-08:
added example frame

[image]

andi34, 2019-09-09:
merged

closed 2019-09-09 by andi34