andi34/photobooth #27: Intruduction : My own Photobooth :)

thymon13, 2019-09-09:
Hello,

I 'm glad to present you my Photobooth.
Thanks to all the people who contribute to the improvement of this project !

I was inspired by other photobooth on the internet. And I liked the one that looks like Instagram.

[image]
[image]

Here it is in action

[image]
[image]
[image]

I recently added the print function.
And I have edit the file "print.php" to add my own frame around the picture (before the last update on this topic)
[image]
[image]

The heart of my baby :) :)
[image]

andi34, 2019-09-09:
Wow, that looks great!
I was thinking about starting a Wiki here instead using the README for everything. Guess we can add a wiki page  where I can add user projects.

Something like that is on my to-do list too. I've a wishlist on Amazon until I've enough money for it (financial situation quite bad atm, should be better in near future).

thymon13, 2019-09-09:
@andi34 
Yes it's great idea to starting a wiki

I have to start another topic. Because with my experience for a lot of event.
I have a lot of demand for enhancement in my mind. :D

flighter18, 2019-09-09:
@thymon13 that looks really amazing!
great work!
mine is not that professional but I really like my bluetooth buzzer :)
[image]
[image]

@andi34 let's start a wiki. that would be great

thymon13, 2019-09-09:
Thanks ! 

Yes it's another great setup, great job @flighter18

thymon13, 2019-09-09:
> @thymon13 that looks really amazing!
> great work!
> mine is not that professional but I really like my bluetooth buzzer :)
> [image]
> [image]
> 
> @andi34 let's start a wiki. that would be great

andi34, 2019-09-09:
@thymon13 I need that buzzer! Where can I get it?

tobiashaas, 2019-09-09:
This is cool! 
http://helge-suess.com/hardware/photo-booth-control/

I Need to take some photos of My Box too, to Show you :)

thymon13, 2019-09-09:
> @thymon13 I need that buzzer! Where can I get it?

@andi34  that's the Flighter18's buzzer :)

flighter18, 2019-09-09:
@thymon13 how do you fix the box on the tripod? 
Could you send some pictures? :)

thymon13, 2019-09-09:
> @thymon13 how do you fix the box on the tripod?
> Could you send some pictures? :)

OK I will do it.
While waiting for the photo. Know that I used a parrabole support

like this exemple 👍 
https://i.ebayimg.com/images/g/4DsAAOSwmLlX-7xn/s-l1600.jpg

flighter18, 2019-09-09:
@andi34 I build it on my own. with this gadgets
https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
https://www.adafruit.com/product/2829

[image]
[image]
[image]

flighter18, 2019-09-09:
> > @thymon13 how do you fix the box on the tripod?
> > Could you send some pictures? :)
> 
> OK I will do it.
> While waiting for the photo. Know that I used a satellite support
> 
> like this exemple :
> https://i.ebayimg.com/images/g/4DsAAOSwmLlX-7xn/s-l1600.jpg

Perfect :) thank you!

thymon13, 2019-09-09:
> Did you make the satellite support by your own? or did you buy it?

Buy it !! :) 

@flighter18  Wireless Buzzer so cool :)

flighter18, 2019-09-09:
> > Did you make the satellite support by your own? or did you buy it?
> 
> Buy it !! :)
> 
> @flighter18 Wireless Buzzer so cool :)

Ok. Thank you :)

flighter18, 2019-09-09:
> @andi34 I build it on my own. with this gadgets
> https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
> https://www.adafruit.com/product/2829
> 

If you push the buzzer shortly the photobox taks one picture. if you press the buzzer long, the photobox starts to create a collage.

thymon13, 2019-09-09:
smart idea.

andi34, 2019-09-09:
Well, Looks like some work and not "doing it quickly" 🙈

thymon13, 2019-09-09:
@andi34
About my multi suggestions to enhancement.
I creat on global subject or one subect by suggestion ?

devnull1975, 2019-09-09:
It looks really nice @thymon13 

Here some Pics of my Photobooth.
There are two pieces. One on the bottom (to insert the plugs for the softboxes) and one on the top.
[image]
[image]
You can connect them in the middle.
[image]

Sliders for the Flash and the cam to adjust them.
[image]

The Printerslot for a better hold.
[image]

The complete Photobooth.
[image]

The flash has a permanent power supply. I make some dummys for the battery and connect them with a 5V carger. It works fine.

andi34, 2019-09-09:
> @andi34
> About my multi suggestions to enhancement.
> I creat one global subject or one subect by suggestion ?

I'll prepare something where you can add it

andi34, 2019-09-09:
> @andi34
> About my multi suggestions to enhancement.
> I creat one global subject or one subect by suggestion ?

https://github.com/andreknieriem/photobooth/issues/68

Nie-Oh, 2019-09-10:
Hello,

Here are some Pictures of my Photobooth.

[image]

The Case is made of IKEA 1x1 Kallax. I use a Touchscreen Monitor directly connected to the Raspi.

[image]

Also, but i have no Images of these Setup, i have an 24" Medion All in One PC and a Mitsubishi 9550 Photoprinter. Every Minute i syncronize the Pictures with WINscp. The Customers can use these PC for printing the Pictures.....

thymon13, 2019-09-10:
@Nie-Oh great !

But is not  andreknieriem's photobooth. Isn't it ?

Nie-Oh, 2019-09-10:
@thymon13 

of course it is. but i realized an Live-Preview with an USB Webcam and motion.

thymon13, 2019-09-11:
@flighter18 
[image]

Nie-Oh, 2019-09-11:
@thymon13 

i use this to mount the Photobooth on the Tripod

https://bit.ly/2ku2z1M

flighter18, 2019-09-13:
> @flighter18
> 
> > @thymon13 how do you fix the box on the tripod?
> > Could you send some pictures? :)
> 
> [image]

@thymon13 

Thanks :)
And how did you the corner on your photobooth? It isn't wood or?

andi34, 2019-09-14:
> @andi34 I build it on my own. with this gadgets
> https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
> https://www.adafruit.com/product/2829
> 

You'd like to sell your buzzer? 😅🙈🤗

thymon13, 2019-09-14:
@flighter18 
foam polyurethane insulating expensive 🤣

flighter18, 2019-09-15:
> > @andi34 I build it on my own. with this gadgets
> > https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
> > https://www.adafruit.com/product/2829
> 
> You'd like to sell your buzzer? 😅🙈🤗

Haha i don't think so :)
But maybe i can build you one?

flighter18, 2019-09-15:
> https://www.instructables.com/id/Instagram-Inspired-DIY-Photo-Booth/

👍 Thank you. will take a look

andi34, 2019-09-18:
> > > @andi34 I build it on my own. with this gadgets
> > > https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
> > > https://www.adafruit.com/product/2829
> > 
> > 
> > You'd like to sell your buzzer? sweat_smilesee_no_evilhugs
> 
> @andi34
> Haha i don't think so :)
> But maybe i can build you one?

Sounds like an idea.
Any idea how much it'll be? (Might have to wait a month or two until I've enough money for it. Financial situation wasn't good for one year now but should be a little better next month.
My stepdaughter now got her own room now too... Lot of bills 🙈)
Guess you're from Germany too?

andi34, 2019-09-26:
@flighter18 thought about it?

flighter18, 2019-09-26:
Sorry i didn't see this message.
i have to calculate it. :) No idea at the moment.

i'm not from germany i am from switzerland :) 

ah yes. I updated my buzzer with a LED strip. It looks fantastic:
take a look if you like:

taking collage:
https://youtu.be/44J-a71J25E 

taking pic:
https://youtu.be/eKONHESbrMw

sorry for the terrible video format. i took those video for instagram :)

plin2, 2019-09-29:
Regarding remote control: Did anyone consider using ESPeasy to build a remote control unit? Some time ago I got myself this gadget (https://de.aliexpress.com/item/32905330340.html?spm=a2g0s.9042311.0.0.27424c4d6qzdZm) which can hold a battery. The only thing missing would be a case and a button.

flighter18, 2019-09-29:
@andi34 
I life near the boarder so i could send the buzzer from germany :)

This are the parts i used for the buzzer:
https://www.play-zone.ch/de/adafruit-feather-32u4-bluefruit-le.html
https://www.play-zone.ch/de/polymer-lithium-ion-lipo-akku-1000mah.html
https://www.bastelgarage.ch/buzzer-taster-rot?search=buzzer
https://www.digitec.ch/de/s1/product/delock-usb-20-a-micro-015m-20-usb-kabel-6089618
http://shop.griederbauteile.ch/product_info.php?cPath=1200_1223&products_id=20111

They costs include postage (from the shop to me)  123.85 CHF

I think 155 Euro + shipping cost would be a fair. 
it would include all the work, parts + buzzer software
What do you think about that?

andi34, 2019-09-29:
Sounds like a fair price. But might be I need 1 or 2 months to have the money for it.
It'll be ok if we talk again once I have the money ready? Private life was a little crazy the last year... Need to watch next month how good things go or how bad things continue 🙈

flighter18, 2019-09-29:
> Sounds like a fair price @flighter18 and thanks a lot for the offer!
> But might be I need 1 or 2 months to have the money for it.
> It'll be ok if we talk again once I have the money ready? Private life was a little crazy the last year... Need to watch next month how good things go or how bad things continue 🙈

No prob! 😄
you can come back when ever you want.

andi34, 2019-10-10:
I am now using a Bluetooth presenter from csl (CSL - Presenter USB 2,4 GHz Fernbedienung für Präsentationen - Hohe Präzision - geeignet für MS PowerPoint Libreoffice Impress Open Office Impress https://www.amazon.de/dp/B07T3BPL5H/ref=cm_sw_r_cp_apa_i_BiYNDbXVBS4FA)
Which uses keys 33 and 34 (page up and down) which works fine on alpha 2.0.0 to trigger photo and collage :)

flighter18, 2019-10-10:
> @flighter18
> 
> > @thymon13 how do you fix the box on the tripod?
> > Could you send some pictures? :)
> 
> [image]

@thymon13 where did you buy the satellite support? :)

thymon13, 2019-10-10:
**Here** : https://www.google.com/maps/@43.2934774,5.4824366,3a,75y,122.14h,99.9t/data=!3m6!1e1!3m4!1sc-cbBCpvIMxS5MpPvr8Jhw!2e0!7i13312!8i6656?hl=fr-FR

🤣 🤣 
Come !!

flighter18, 2019-10-10:
> @flighter18
> **Here** : https://www.google.com/maps/@43.2934774,5.4824366,3a,75y,122.14h,99.9t/data=!3m6!1e1!3m4!1sc-cbBCpvIMxS5MpPvr8Jhw!2e0!7i13312!8i6656?hl=fr-FR
> 
> 🤣 🤣
> Come !!

@thymon13 
😆 😆 thank you 👍 will be there tomorrow

flighter18, 2019-10-10:
@thymon13 actually there is a shop near Mulhouse. So not far from my home. 
Maybe i can get it there 😃 

Do you have a link to this "satellite support" on leroymerlin.fr? I couldn't find it because everything is in french 😅

firesta666, 2019-10-10:
How it works with the Bluetooth presenter and the config to Pins? 

Thanks for help

andi34, 2019-10-10:
@firesta666 in v2.0.0-alpha.
Set config for photo key: 33 or 34 (Page Up and Page down on Presenter)

To also use a GPIO buzzer:
For GPIO use the Python script in root of the source and modify the script to send accesskey 33/34

andi34, 2019-10-10:
Connect by Presenter with your pi,
Open https://keycode.info ,
Press a button on your presenter

```
// specify key id to use that key to take a picture or collage (e.g. 13 is the enter key)
// use for example https://keycode.info to get the key code
$config['photo_key'] = null;
$config['collage_key'] = null;
```

thymon13, 2019-10-11:
@flighter18 

Indeed, I can not find it anymore.
😢

firesta666, 2019-10-11:
@Andi34: Thanks for information.

Question: Where can i found the Alpha 2.0.0-alpha?

andi34, 2019-10-11:
@firesta666 follow https://github.com/andreknieriem/photobooth/wiki/Update-Photobooth

flighter18, 2019-10-14:
> @flighter18
> 
> Indeed, I can not find it anymore.

@thymon13 sad story :cry: :laughing:

flighter18, 2019-11-18:
@thymon13 look my new photobooth :smile: 

[image]

irqnet, 2020-01-19:
> @thymon13 look my new photobooth 
> 
> [image]

Is the light always on, or how do you control it while taking a picture?

flighter18, 2020-01-20:
@irqnet 
no it isn't always on. 
The LED stripe does like a countdown while the the user is seeing the countdown on the display and finally the stripe does a pattern while the finished pictures is showing.

michi-93, 2020-06-14:
Hi! 
i want to show you my version of the photobooth. I designed the box on the 3D software and a fried of mine who owns a CNC wood milling machine has cut it for me. 
[image]
This is the digital version.
[image]
This is my real version

andi34, 2020-08-09:
User collection can be found here: http://andi34.github.io/userprojects.html

47photography, 2021-07-07:
Hello @all!

Here is my 3. Version of the Photobooth and i think this is my final one.

Do anyone knows how i can watermark the photos? Not printing only digital.

[image]
[image]
[image]

andi34, 2021-07-07:
To watermark you can use a "frame" which adds a watermark (instead being a frame itself) or use the text on image function.

47photography, 2021-07-07:
@andi34 where i find text on image? i only find the option for "printing".

andi34, 2021-07-07:
You need latest Photobooth v3 installed

47photography, 2021-07-08:
> 
> 
> You need latest Photobooth v3 installed

Okay. I found it but i couldnt install the update. 

git checkout origin/stable3

Here i get one error about "nothing found".

Is there option to contact you by email or so? maybe you can help to me. Im from Germany, so we can talk in german too.

[2 more replies]