andi34/photobooth #35: Make it possible to change/Add Style in the Admin Panel with Selection.

tobiashaas, 2019-10-02:
Hello everybody,

I am currently testing a bit, but I can not get what I want ...

I would like be able to select the theme via a selection box in the Admin Panel, so far it is only possible to activate or deactivate the Bluegray Theme.

Unfortunately I do not know how to link my code here, so I'm doing it the classic way ;-)
https://github.com/tobiashaas/photobooth/tree/dev

My changes can be found in the following files
- index.php
- theme.js
- /admin/index.php

Best Regards 
Tobi

andi34, 2019-10-02:
https://github.com/andi34/photobooth/commit/b10248c9f5b526ae1f88b2c79ecae53ecf30a129

Will be added once the refactoring from sualko is done

closed 2019-10-02 by andi34

andi34, 2019-10-02:
Sualko hat das ganze überarbeitet: https://github.com/sualko/photobooth/commit/4ef2aa3f9312f7ce80a30c442f75afd348135194

Sollte demnächst dann reinwandern.
Muss dann nur in die config.php:

```
    'yellow' => [
        'primary' => '#fcff00'',
        'secondary' => '#2e535e'',
        'font' => '#f0f6f7''',
    ]
```

Im theme.js:
```
if (config.color_theme === 'yellow-bg') {
    $('#wrapper').addClass('yellow-bg');
}
```

Im style.sass
```
&.yellow-bg {
    background-image: url(../img/bg_yellow.jpg);
  }
```
Hast du das bg_yellow für mich? Ist bei dir nicht mit drin gewesen.

tobiashaas, 2019-10-02:
Klar, das kann ich dir gerne geben.
Das Yellow war aber nur mal so just4fun :-D 
[image]

andi34, 2019-10-02:
Kannst du auch gerne noch optimieren oder anhand der Infos was anderes feines vorbereiten ;)

tobiashaas, 2019-10-02:
Ich bin gerade noch dabei eine etwas andere Optik zu kreieren :-D 
Wenn man den Hintergrund für eine Hochzeit personalisiert, dann hat man den Text und die Buttons auf dem Gesicht liegen, was nicht schön ist... Siehe Index_test.php in meinem Branch

Aber mal schauen ob ich das hinbekomme ;-)

War auf jeden Fall schon mal nicht so falsch mit dem Yellow Theme, hab da schon an den richtigen Stellen rumgebastelt :-D

tobiashaas, 2019-10-02:
Was ich mich noch gefragt habe...
Theme.js 
`if (config.color_theme === 'yellow') {
    $('#wrapper').addClass('yellow-bg');
}`

Wäre es da nicht sinnvoller wenn man das in die Haupt Index.php einfügt?
So könnte man den Hintergrund über das Admin Panel austauschen.

Zb. 
`Body style="background: url('/resources/img/bg.jpg') center center no-repeat;"`

Da könnte man doch sicher die url per PHP austauschen oder nicht? 

Grüße ;-)

andi34, 2019-10-02:
Gute Frage. Vllt sobald die Option drin ist als Enhancement issue aufmachen. Mal abwarten was Klaus da noch zaubert.

tobiashaas, 2019-10-02:
> Sualko hat das ganze überarbeitet: [sualko@4ef2aa3](https://github.com/sualko/photobooth/commit/4ef2aa3f9312f7ce80a30c442f75afd348135194)
> 
> Sollte demnächst dann reinwandern.
> Muss dann nur in die config.php:
> 
> ```
>     'yellow' => [
>         'primary' => '#fcff00',
>         'secondary' => '#2e535e',
>         'font' => '#f0f6f7',
>     ]
> ```
> 
> Im theme.js:
> 
> ```
> if (config.color_theme === 'yellow') {
>     $('#wrapper').addClass('yellow-bg');
> }
> ```
> 
> Im style.scss
> 
> ```
> &.yellow-bg {
>     background-image: url(../img/bg_yellow.jpg);
>   }
> ```
> 
> Muss natürlich die style.css dann noch neu erstellt werden.
> 
> Hast du das bg_yellow für mich? Ist bei dir nicht mit drin gewesen.

Hat geklappt! ;-)
Muss es nun nur noch anpassen
Danke dir!

andi34, 2019-10-03:
> Ich bin gerade noch dabei eine etwas andere Optik zu kreieren :-D
> Wenn man den Hintergrund für eine Hochzeit personalisiert, dann hat man den Text und die Buttons auf dem Gesicht liegen, was nicht [...]

https://github.com/sualko/photobooth/commit/4c2e6b0a2c8b16f817c49b585b75b0234be325c4

;)