andi34/photobooth #42: No Foto taken with "Live-Preview enabled during Countdown"

eierbatz, 2020-03-05:
**Describe the bug**
Hallo, danke für die Anwendung an den Entwickler , super Sache.

Mein Problem:
Ich habe Photobooth2.2.1.zip von den Isues kopiert und auf mein Rasp kopiert. Funktioniert soweit.
Livewallpaper funktioniert über die Tabletcam bei dem Homescreen und auch bei dem Countdown. Aber nach dem Cheeese wird kein Bild aufgenommen und ein Fehler ausgegeben.
Deaktiviere ich das Livewallpaper während dem Countdown funktioniert wieder die Aufnahme.

Ich würde gerne das Livebild während dem Countdown verwenden, damit die Fotomodelle sich in der Zeit korrekt ausrichten können.

Kann ich irgendwelche Logfiles genererieren oder auslesen, die mir hinweise geben was bei der Fotoaufnahme schief läuft?

Danke

**To Reproduce**
Vorschau während Countdown an, Foto wird nicht erstellt
Vorschau während Coutndown aus, Foto wird erstellt.

**Screenshots**
könnte ich nachreichen, zur Zeit nicht möglich.

**Environment (please complete the following information):**
Tablet, mit Android 7, Chrome, für die Oberfläche
Raspberry PI 3b+ mit Raspbian CLI für die Verarbeitung

**Photobooth**
- Did it work before? 
yes
- Which version of Photobooth are you using?
2.2.1

andi34, 2020-03-05:
Welche Optionen zur Aufnahme nutzt du?
Gphoto?
Wenn du eine Raspberry Camera Nutzt funktioniert die Aufnahme nicht wenn Motion aufgesetzt ist.

Am Besten bitte deine my.config.inc.php Posten damit ich es ggf. nachstellen kann.

eierbatz, 2020-03-05:
Hallo

Ich nutze Gphoto um ein Bild von einer externen Kamera (Canon) per USB zu Capturen. Die Vorschau läuft soweit und verwendet die Tabletkamera, (Android App um die Kamera als Webcam zu nutzen mit url(http://cameraIP:8080) )

Die Fehlermeldung ist sehr generell gehalten :-) Fehler bei der Aufnahme. Kann man mehr Text erwarten, wenn der Entwicklermodus im Adminpanel angeschaltet ist?
Komme gerade nich an das System ran :-) Kann heute Nachmittag die Config posten.

andi34, 2020-03-05:
Unabhängig von Entwickler Modus, "Fehlermeldungen anzeigen" aktiviert lassen.

Kommt dann eins der folgenden:
- Gphoto returned with an error code
- File was not created
- No style provided
- No or invalid collage number provided
- Collage consists only of 4 pictures
- Invalid photo style provided

andi34, 2020-03-05:
Ich hab's gerade Mal schnell mit IP Webcam aus dem Playstore versucht (https://play.google.com/store/apps/details?id=com.pas.webcam) und zumindest im Dev Mode keine Probleme die Vorschau zu sehen und Bild sozusagen "auszulösen".

Poste doch einfach einmal deine my.config.inc.php und ich schaue Mal ob ich was sehe.
Bitte auch die Fehlermeldung die erscheint wenn du auslöst.
Ansonsten schau Mal am PC im Browser über F12 in die Konsol ob dort was drin steht.

andi34, 2020-03-05:
Eine Idee habe ich bezueglich falschem Setup:

Beides aktiviert:
- Vorschau aus der Gerätekamera anzeigen
- Gerätekamera zur Bildaufnahme verwenden

Heisst: du siehst die Vorschau aus z.B. deiner Tablet Kamera wenn du dort die Booth aufrufst, dieses Bild wird aufgenommen. Hier wird gphoto dann **NICHT** genutzt. 
Heisst aber auch: Sollte kein Stream erzeugt werden, z.B. weil dein Geraet keine Kamera hat, so wird ein fehler ausgespuckt. Siehe: https://github.com/andi34/photobooth/commit/26b85de1f431193e0dee01db658b00d3a29ed78c
(vorher war das Resultat der Aufnahme einfach ein schwarzes Bild ohne Fehlerhinweis)

Hast du diese beiden Optionen aktiviert siehst du **NICHT** deine Tablet Kamera auf einem anderen Geraet wo die Booth aufgerufen wird. Hat das Geraet eine eigene Kamera siehst du diese stattdessen, ansonsten eben nicht und die Aufnahme ist nicht moeglich.

**Willst du z.B. deine Tablet Kamera auf allen Geraeten sehen wo die Booth genutzt/aufgerufen wird:**
Du musst "Vorschau von URL" aktivieren und von deinem Tablet einen Stream bereitstellen, z.B. mit der vorher erwaehnten App. 
Als "Vorschau-URL" dann entsprechend im Adminpanel eintragen, bei mir `url(http://192.168.0.94:8080/video)` .

Folgende Optionen dann bitte **nicht** aktivieren:
- Vorschau aus der Gerätekamera anzeigen
- Gerätekamera zur Bildaufnahme verwenden

andi34, 2020-03-05:
https://github.com/andi34/photobooth/wiki/FAQ#how-to-use-a-live-stream-as-background-at-countdown

andi34, 2020-03-08:
@eierbatz kann ich das Issue schließen?

eierbatz, 2020-03-08:
Hallo, danke für deine Mühe.
Komischerweise bekomme ich es nicht mehr hin, dass es.nicht mehr geht. Denke es lag an meinen Aufnahmebefehl, den ich angepasst habe. Nicht captureimage sondern triggercapture... ein längeres wait and download hat scheinbar das Problem gelöst.

andi34, 2020-03-08:
Ok, Dann schließe ich das Issue.
Danke für die Rückmeldung.

closed 2020-03-08 by andi34