andi34/photobooth #46: Implement fullscreen button for android tablets

md-20, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
On my android tablet status bar and tabs are shown when i open the photobooth-page. I cannot open fullscreen mode.
But when I implement following simple code in index.php it works, maybe you could implement it as an option?:
`<button id="goFS">Go fullscreen</button>
<script>
  var goFS = document.getElementById("goFS");
  goFS.addEventListener("click", function() {
      document.body.requestFullscreen();
  }, false);
</script>
`
Source: [https://developers.google.com/web/fundamentals/native-hardware/fullscreen](url)
Thank you for your amazing work with this photobooth!

andi34, 2020-04-01:
I'll take a look at it next days :)

andi34, 2020-04-06:
Good idea where to place the button?

[image]

[image]

andi34, 2020-04-06:
[image]

i think it's looking quite ok now.

what do you think?

md-20, 2020-04-06:
Yes it looks awesome. Will it Look in landscape mode similiar?-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

andi34, 2020-04-06:
Depends on your display size, on 6 inch:

**Landscape**
[image]

[image]

**Portrait**
[image]

[image]

md-20, 2020-04-06:
Awesome, you rock! Do you implement it in options to enable and disable for those who want do decide if they need that option?

Oh and i wonder if the button could disappear when already in fullscreen? Maybe its not easy to implement?
 I think i can switch to normal mode by hitting the back button. But then it must reappear when going back to normal mode. 
No, i think the button can stay there :) So, if i want the button to disappear i can disable the option in admin panel when already in fullscreen, that should do it!

Thank you!

andi34, 2020-04-06:
It's easy to hide it if you are in fullscreen, but if you exit fullscreen without reloading the page the button would still be gone.

andi34, 2020-04-06:
```
diff --git a/resources/js/core.js b/resources/js/core.js
index 8cf8190..0a6b3b7 100644
--- a/resources/js/core.js
+++ b/resources/js/core.js
@@ -653,6 +653,7 @@ const photoBooth = (function () {
         e.preventDefault();
         if (!document.fullscreenElement) {
             document.body.requestFullscreen();
+            $('#fs-button').hide();
         } else {
             if (document.fullscreenElement) {
                 document.exitFullscreen();
```

andi34, 2020-04-06:
Ready to use build:
[photobooth-SNAPSHOT-20200406.zip](https://github.com/andi34/photobooth/files/4439215/photobooth-SNAPSHOT-20200406.zip)

md-20, 2020-04-06:
> 
> 
> It's easy to hide it if you are in fullscreen, but if you exit fullscreen without reloading the page the button would still be gone.

i think you could implement a toast message on pressing the button like " Please reload the page when you turn back to normal mode". Other option: Just writing "Please reload the page when you turn back to normal mode" after the option in admin panel

md-20, 2020-04-06:
> 
> 
> Ready to use build:
> [photobooth-SNAPSHOT-20200406.zip](https://github.com/andi34/photobooth/files/4439215/photobooth-SNAPSHOT-20200406.zip)

Great! Sorry, i have a look later again, im busy now

andi34, 2020-04-06:
> i think you could implement a toast message on pressing the button like " Please reload the page when you turn back to normal mode". Other option: Just writing "Please reload the page when you turn [...]

I'll think about it. But if you have time feel free to implement and send a PR.

md-20, 2020-04-06:
It looks great and works nicely, thank you! Well done! I think you can keep it that way, the button doesn't matter, its pretty!
[image]

andi34, 2020-04-07:
Great!
Let me know if you find any issues, the package noch also used a different translation library which supports a fallback if a translation is missing. Also i18n-translator can be used to translate Photobooth.

For reference https://github.com/andi34/photobooth/issues/41

closed 2020-04-11 by andi34