andi34/photobooth #47: iPad 2 compatibility

andi34, 2020-04-05:
**Is your feature request related to a problem? Please describe.**
Safari Browser on iPad 2 in not compatible with es6.

**Ready to use packages**
- [ipad2-photobooth-2.2.1.zip](https://github.com/andi34/photobooth/files/4433493/ipad2-photobooth-2.2.1.zip)
- [ipad2-photobooth-2.2.1.tar.gz](https://github.com/andi34/photobooth/files/4433492/ipad2-photobooth-2.2.1.tar.gz)

**Source changes for reference**
https://github.com/andi34/photobooth/commit/98e9055654486046e9af9c4a6e9bcd01d13f1fec

andi34, 2020-05-10:
**Ready to use packages**
- [photobooth-2.2.2.zip](https://github.com/andi34/photobooth/files/4606339/photobooth-2.2.2.zip)
- [photobooth-2.2.2.tar.gz](https://github.com/andi34/photobooth/files/4606340/photobooth-2.2.2.tar.gz)

Added bugfixes and new options from v2.3

andi34, 2020-06-06:
**Ready to use packages**
- [photobooth-2.2.3.zip](https://github.com/andi34/photobooth/files/4739854/photobooth-2.2.3.zip)
- [photobooth-2.2.3.tar.gz](https://github.com/andi34/photobooth/files/4739856/photobooth-2.2.3.tar.gz)

Based on: https://github.com/andi34/photobooth/commits/860070261a80da2e1ebe4b681a358aac9c853ac5

Includes new options from v2.4.0

andi34, 2020-07-02:
**Ready to use packages**
- [photobooth-2.2.4.zip](https://github.com/andi34/photobooth/files/4865261/photobooth-2.2.4.zip)
- [photobooth-2.2.4.tar.gz](https://github.com/andi34/photobooth/files/4865262/photobooth-2.2.4.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/da617bcc002318f38652ff635bce24ae33d11d57

  Includes new options from v2.5.0

andi34, 2020-07-09:
**Ready to use packages**
- [photobooth-2.2.5.zip](https://github.com/andi34/photobooth/files/4899068/photobooth-2.2.5.zip)
- [photobooth-2.2.5.tar.gz](https://github.com/andi34/photobooth/files/4899065/photobooth-2.2.5.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/fb1ab971fc99dab5ef26b2deb75168902992dd7b

  Includes new options from v2.6.0

**New options**
- Automatically reload Photobooth if an error occurs while taking a photo/collage (enabled by default)

**Bugfixes**
- Fix FC on Standalone Gallery if a keycode is defined to take a photo/collage
- Close gallery if a keycode is defined to take a photo/collage

**General changes**
- update PHPMailer to latest version

andi34, 2020-07-16:
**Ready to use packages**
- [photobooth-2.2.6.zip](https://github.com/andi34/photobooth/files/4930716/photobooth-2.2.6.zip)
- [photobooth-2.2.6.tar.gz](https://github.com/andi34/photobooth/files/4930715/photobooth-2.2.6.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/88f29953cc86a5139a04db710cf9ccb18155e9e7

    Includes changes from v2.6.1:  
    **Bugfixes**
    - Fix video--sensor canvas
    - Update Style for 5inch Display (800x480px)
    - Attempt to fix taking Pictures and Collage via defined keys

    **General changes**
    - Add offline FAQ, access directly via [http://localhost/manual/faq.html](http://localhost/manual/faq.html)
    - Update jQuery to v3.5.1

andi34, 2020-07-29:
**Ready to use packages**
- [photobooth-2.2.7-2.zip](https://github.com/andi34/photobooth/files/4996557/photobooth-2.2.7-2.zip)
- [photobooth-2.2.7-2.tar.gz](https://github.com/andi34/photobooth/files/4996556/photobooth-2.2.7-2.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/de67782876accd2ebdf7a5c2c39c3fcfad6b673d

    Includes bugfixes from v2.7.2

- [photobooth-2.2.7-1.zip](https://github.com/andi34/photobooth/files/4975180/photobooth-2.2.7-1.zip)
- [photobooth-2.2.7-1.tar.gz](https://github.com/andi34/photobooth/files/4975178/photobooth-2.2.7-1.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/095bb9c2cab735cf65eccee0dde4ad329f05037c

    Includes bugfixes from v2.7.1

- [photobooth-2.2.7.zip](https://github.com/andi34/photobooth/files/4961573/photobooth-2.2.7.zip)
- [photobooth-2.2.7.tar.gz](https://github.com/andi34/photobooth/files/4961570/photobooth-2.2.7.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/25acb772b8d792454eaa84dc95e837898a701f3f

    Includes changes from v2.7.0

- [photobooth-2.2.6.zip](https://github.com/andi34/photobooth/files/4930716/photobooth-2.2.6.zip)
- [photobooth-2.2.6.tar.gz](https://github.com/andi34/photobooth/files/4930715/photobooth-2.2.6.tar.gz)

  - Based on: https://github.com/andi34/photobooth/commits/88f29953cc86a5139a04db710cf9ccb18155e9e7

andi34, 2020-08-01:
Someone able to test the following package while accessing Photobooth from ipad2?

- [photobooth-SNAPSHOT.20200801085310032.zip](https://github.com/andi34/photobooth/files/5010040/photobooth-SNAPSHOT.20200801085310032.zip)

- [photobooth-SNAPSHOT.20200801085310032.tar.gz](https://github.com/andi34/photobooth/files/5010041/photobooth-SNAPSHOT.20200801085310032.tar.gz)

jacques42, 2020-08-01:
I'm on it - anything particular you want me to pay attention to ? 

btw happy to help maintain this package, let me know how I can help more.

andi34, 2020-08-01:
@jacques42 Would be good to know if it works without issues. If yes we can skip the seperate branch and use one main branch again :) it's based on https://github.com/andi34/photobooth/commits/add-build-pipeline

Andreas Remdt added a transpiler :)

jacques42, 2020-08-01:
@andi34 it's not working out of the box. Text content / translations do not show, In buttons I see empty text fields . On the admin screen I get to see the component text placeholder name (again, not even english). 

I may be wrong but looks like as if the Translator class crashes on iPad2. I will do some debugging and see if I can find out what's going on.

andi34, 2020-08-02:
@jacques42 thanks for testing!
Could you please enable dev mode and check the browser console for errors (press F12 ) ?

andi34, 2020-08-02:
@jacques42 you can also join https://t.me/PhotoboothGroup if you like to talk about the issue.

jacques42, 2020-08-02:
I found that Safari 9 throws the following error from `translator.min.js`
```
ReferenceError: Can't find variable: fetch 
```
After a bit of digging, it looks to me that the root cause may be iOS 9 / Safari 9 did not yet implement the fetch() API ([Fetch API compatibility matrix](https://developer.mozilla.org/de/docs/Web/API/Fetch_API#Browserkompatibilit%C3%A4t))

There is a polyfill available that from a very first glance could be a solution to backwards compatibility - [fetch](https://github.com/github/fetch)

The best way I found to debug iPad2 in the Photobooth is actually via USB connected to the Pi it self, and using the instructions right here: [remotedebug-ios-webkit-adapter](https://github.com/RemoteDebug/remotedebug-ios-webkit-adapter)

andi34, 2020-08-03:
Another one to test which hopefully fixes it
- [photobooth-SNAPSHOT.20200803181225859.zip](https://github.com/andi34/photobooth/files/5017577/photobooth-SNAPSHOT.20200803181225859.zip)

- [photobooth-SNAPSHOT.20200803181225859.tar.gz](https://github.com/andi34/photobooth/files/5017578/photobooth-SNAPSHOT.20200803181225859.tar.gz)

jacques42, 2020-08-03:
@andi34 @andreasremdt I've tested the [whatwg-fetch](https://github.com/github/fetch) polyfill in `simple-translator` as provided in above snapshot - works great on iOS 9 / iPad2, thanks a  million. 

I found one more small thing to be adjusted, in order to make `simple-translator` work with Photobooth on iPad2. 

There seems to be an [Javascript ES6 TypeError in Safari & IE](https://stackoverflow.com/questions/45971119/javascript-es6-typeerror-in-safari-ie) when iterating over a NodeList object using `forEach`. Along that comment, I tried the following change in file `translate.js` member function `translatePageTo`:

```
<                     elements.forEach(function(element) {
---
>                     Array.from(elements).forEach(function(element) {
```

which then made Photobooth translations work perfectly well on iPad2. 

I tried to provide a simple PR but was not able to find the repo / branch where the backward compatible code base resides which is included in the snapshot (i.e. no more use of arrow function). If you are ok to take it from here would be great or let me know where to look and I am happy to provide a PR for this change.

Once more thanks for all the support and your willingness to go beyond, and to make it work.

andreasremdt, 2020-08-04:
Hey @jacques42, thanks for the interesting feedback. Indeed, iterating over a NodeList using `forEach` is not something older browsers support. Since this is a rather small change, I will update Simple Translator today and then update the PR with the fixed version. That should fix it

andi34, 2020-08-04:
@jacques42 [photobooth-2020721710.zip](https://github.com/andi34/photobooth/files/5022886/photobooth-2.8.0-alpha-2020721710.zip)
[photobooth-2020721710.tar.gz](https://github.com/andi34/photobooth/files/5022887/photobooth-2.8.0-alpha-2020721710.tar.gz)

jacques42, 2020-08-04:
Yeay - works like a charme 🥇 thank you so much for going that extra mile !

There is two small changes please to be considered - have a look at [commit 0e9415e77be6e558116c6d02510fd769b45c8e98 ](https://github.com/jacques42/photobooth/commit/0e9415e77be6e558116c6d02510fd769b45c8e98) 

First change is to make translations also work on the admin page - adjusted URL to the whatwg-fetch polyfill module

On the second change, interestingly with changing the default URL references for the background picture, that background picture does load on my iPad2 very nicely. However and to my surprise it does not load on my desktop (Firefox, Chrome). Desktop browsers don't throw an error on the console but also don't show the background picture. Not sure what's going on there.

andi34, 2020-08-05:
Thanks for the finding!
About the background:
You've tries on an incognitotab? Sometimes the browser cache can cause some trouble.

andi34, 2020-08-05:
Fixed the whatwg-fetch path also on other pages:
[photobooth-SNAPSHOT.20200805035943339.zip](https://github.com/andi34/photobooth/files/5026035/photobooth-SNAPSHOT.20200805035943339.zip)

andi34, 2020-08-08:
neues update, "whatwg-fetch" auf version 3.4.0 upgedated.
[photobooth-SNAPSHOT.20200808205436739.zip](https://github.com/andi34/photobooth/files/5046199/photobooth-SNAPSHOT.20200808205436739.zip)

andi34, 2020-08-13:
https://github.com/andi34/photobooth/releases/tag/v2.8.0

closed 2020-08-13 by andi34