andi34/photobooth #51: Language Files are not detected if photobox is not installed in home-dir

danielh84, 2020-04-12:
**Describe the bug**
i installed the fotobox in the folder /photobox/ instead of /. 

All files are loaded instead of

de.json
en.json

They got a 404 because they are searched in /resources/lang/ instead of /photobox/resources/lang/

**To Reproduce**
Steps to reproduce the behavior:
1. Install photobox in a subdir
2. Open Admin View or Main Site

**Expected behavior**
All files are loaded well

**Environment (please complete the following information):**
 - Version 2.3.0 (develop Branch)

**Photobooth**
- Did it work before? 2.2.1-alpha works well

andi34, 2020-04-12:
@danielh84 thanks for reporting the bug!

Could you please also test if https://github.com/andi34/photobooth/commit/6203d6e2306c51a8c1d0e06d1709712f302bb9f8 fixes the problem for you? I've no trouble here now.

danielh84, 2020-04-12:
The main-site works fine.... 

Admin site gets an errror. It wants to find the files in

/photobooth/admin/resources/lang/ and not 
/photobooth/resources/lang/

andi34, 2020-04-12:
ok, thanks. i see... :disappointed: 
let me see what i can do. might find some more time later, else tomorrow to fix it.

andi34, 2020-04-12:
Latest dev branch should work now, only not happy the way to get it working.

andi34, 2020-04-13:
Fix merged and build published.
i'll try to find a better solution for future releases.

closed 2020-04-13 by andi34