andi34/photobooth #64: [Help needed] update translations

andi34, 2020-06-07:
Hey together!

Photobooth now reached v2.4.0 and a lot of translation are missing for French, Greek and Spanish. It would be great if someone is able to help on completing the translation on those languages.

For sure, I'll also accept new languages 🙂

Thanks to i18n-editor it's easy now to translate the project - more about it here:  [How to update or add translations?](https://github.com/andi34/photobooth/wiki/FAQ#how-to-update-or-add-translations)

Translation done on french in the past: @thymon13
Translations done for Greek in the past: @fossil74

You're able to help?

stroudn1, 2020-06-22:
Hi! I'd love to help with Spanish translation if at all possible. :)

andi34, 2020-06-23:
@stroudn1 that would be great! You can follow the instructions mentioned in the Wiki
 I18n-editor works fine and makes it easy to translate.

andi34, 2020-09-12:
Photobooth joined Crowdin as localization manager!

Translating Photobooth is easier than ever before!
Create an account on Crowdin and apply as translator on  https://crowdin.com/project/photobooth

Your language is missing? Let me know and I'll add it to the project 😊

Best regards

Andi

andi34, 2021-02-24:
Translation currently not possible because of a big non-public source rework for next release. No ETA. Translation on crowdin will be possible again as soon as possible.

andi34, 2021-02-28:
Translation possible again. Language files have been updated for upcoming v3.0.0.

andi34, 2021-03-13:
@thymon13 you like to take a look at the french translations for v3?

andi34, 2022-01-30:
## [ℹ️ Information] remove of unmaintained languages

Hey together!
Our Photobooth project got a lot of changes in the past and a big refactoring from v2 to v3. A lot of new settings have been added and we're happy that most of the user wishes have made it's way into the project.

I've noticed, that most languages haven't received any updates for a long time now. For the upcoming release we're going to remove languages without Updates, for latest development release they get removed later today already.
Maybe you like to take a look at the open strings for your language to keep your language supported/added back?

Languages which have been supported in the past by Photobooth but being disabled until they get maintained:
- French
- Polish
- Greek
- Spanish

New translation will make it's way into latest Photobooth development version faster from now! I've setup some GitHub action workflows:
- source file on Crowdin automatically gets updated once the English file changes
- translations will be pushed in a Pull Request weekly to get into the project (depending on the activity on Crowdin we might choose a different schedule)

If you're wondering what have changed between the versions and what's going to change on next release:
An up to date Changelog can be found at https://github.com/andi34/photobooth/blob/wiki/Changelog.md
Upcoming changes have been added to the development version already.

What we can reveal: there's some work in progress for a new Gphoto Integration via Python which allows a better experience while using Gphoto for preview at Countdown!

We hope you're enjoying the project! Don't hesitate to contact me if you're having any questions!

Best regards

Andi