andi34/photobooth #73: Gallery: Close opened picture if photo/collage is triggered

andi34, 2020-07-04:
**Describe the bug**
If a picture is opened in gallery and a picture is taken via buzzer/keycode the countdown and everything following is not visible.

**To Reproduce**
Define a keycode to trigger a picture via admin panel.
Take a picture.
Open gallery and click on a thumbnail to view the picture.
Trigger a photo via defined keycode.

**Environment (please complete the following information):**
 - OS: Raspbian OS

**Photobooth**
- Did it work before? 
No
- Which version of Photobooth are you using?
2.5.0

**Additional context**
Picture is a pswp element. We need to hide it if a picture is taken and make it visible again.

andi34, 2020-07-04:
@sualko you've an idea how to fix that?

sualko, 2020-07-06:
Hi Andy, currently I can't help because I didn't use it for quite some time, because I'm busy with other things.

sualko, 2020-07-06:
Maybe a quick idea: You should register an event handler in https://github.com/andi34/photobooth/blob/dev/resources/js/photoinit.js which is listening for the trigger key and call `gallery.close()`. Your linked commits will result in a huge mess. Maybe you can also make `gallery` globally callable, but that would be the nasty way.

andi34, 2020-07-06:
It's the `pswp` class which has the picture opened.
I somehow need to use the `pswp.close()` from https://github.com/andi34/PhotoSwipe/blob/master/dist/photoswipe-ui-default.js#L734

But didn't got it working so I've had to remove all the classes this way (else it'll add the other classes a second time if a picture is opened)....

andi34, 2020-07-06:
https://github.com/andi34/photobooth/commit/81aab3565292176fcc1e9a95a7853a2d970ea084

Ok, this works too 🙈
Thanks @sualko !

andi34, 2020-07-07:
Well... Seems like my cache had a false positive result. Does not work this way... Need to play around again

closed 2020-07-09 by andi34