andi34/photobooth #79: Question: Live View not working

pichapee, 2020-07-13:
Hey there,

I'm trying to get the Live View working for a few days now and don't know what to do right now.
My knowledge of linux is very basic and i couldn't find any more sources to look up my question,  so please forgive me :)

What i tried so far:

-Installed Master-Branch from andreknieriem, from script. Setup as in Wiki. Everything works accept live view.
-Installed your branch (andi34) from script. (Looked better in terms of live-view). Setup everything like mentioned in the wiki.
Still no live view at all .
-Tested live view from bash `gphoto2 --capture-movie --stdout | ffmpeg -re -i pipe:0 -listen 1 -f mjpeg http://localhost:8080/feed.jpg` . <- got an output, so the camera should work in general.
-Accessed site form different devices. Android Smartphone, Ipad, Localhost (raspberry pi). No Preview form DSLR AND/OR the devices cams i used.
-Tried every possible configuration in the admin settings

I am using:
-RaspberryPi 3b+. Newest Raspberry Pi OS, Version: May 2020
-Canon EOS 1000D (Image Capture, Trigger Capture, Liveview, Configuration). Live View is on. Connected via USB

Thanks!

andi34, 2020-07-13:
> -Tested live view from bash `gphoto2 --capture-movie --stdout | ffmpeg -re -i pipe:0 -listen 1 -f mjpeg http://localhost:8080/feed.jpg` . <- got an output, so the camera should work in general.

Currently there's no working way known to me, to use gphoto for preview.
But in general: If you generate a stream, you could define the stream via admin panel for Preview from URL at countdown for example - this would be the same on all devices you access Photobooth from.
- Admin panel config "Preview from URL": `true`
- Admin panel config "Preview-URL" example (add needed IP address instead): `url(http://127.0.0.1:8081)`

> -Accessed site form different devices. Android Smartphone, Ipad, Localhost (raspberry pi). No Preview form DSLR AND/OR the devices cams i used.
> -Tried every possible configuration in the admin settings

If you access Photobooth from a different device, you need to define your IP as exception inside chrome or setup https access to Photobooth. If you use the preview from device cam, "device cam" is always the cam of the device you access Photobooth from.
How to define the exception: https://medium.com/@Carmichaelize/enabling-the-microphone-camera-in-chrome-for-local-unsecure-origins-9c90c3149339
- Admin panel config "See preview by device cam": `true`

pichapee, 2020-07-13:
> Currently there's no working way known to me, to use gphoto for preview.

Okay, didn't know that, thought there would be an "easy" way to enable it.

> If you access Photobooth from a different device, you need to define your IP as exception inside chrome or setup https access to Photobooth. If you use the preview from device cam, "device cam" is always the cam of the device you access Photobooth from.

Actually entered the wrong ip-adress * **facepalm** *. Thanks for reminding me. Cam from Android device is now working.

Looks like im going to use the cam from the tablet.

Thanks for your help!

closed 2020-07-20 by andi34