andi34/photobooth #81: Collage: invaild file provided

nocomment-bln, 2020-07-24:
Hallo Andi34,

die Collagen Funktion funktioniert bei mir leider nicht.

Es werden alle 4 Fotos ausgelöst und beim zusammenrechnen kommt dann das:

[image]

Hast du eine Idee woran es liegen könnte ?

Installiert ist: 2.7.0 auf nem Raspi 4

andi34, 2020-07-24:
Schau ich mir nach dem Aufstehen an.
Hast du Mal den Browser cache geleert? Oder über ein Privates/Incognito Fenster geöffnet und versucht?
Ist über F12 was in der Konsole noch sichtbar?

andi34, 2020-07-25:
Problem gefunden. Ich schaue gerade nach einer Loesung

closed 2020-07-25 by andi34

andi34, 2020-07-25:
v2.7.1 behebt das Problem und ist ab jetzt verfuegbar.
Vielen Dank fuers melden des Fehlers!

nocomment-bln, 2020-07-25:
Danke Andi! Super schneller Support! Gute Arbeit!