andi34/photobooth #83: Preview from gphoto

andi34, 2020-07-25:
**Is your feature request related to a problem? Please describe.**
No and yes. It would be good to be able to only use the main camera also to generate a preview.
Currently we need a 2nd camera to use it or it's stream.

**WIP:**
https://github.com/andi34/photobooth/tree/feature/gphoto-preview

**Feel free to play around with it**

1. If you have installed Photobooth via git from my fork:
    Open a Terminal and navigate into your installation folder, by default /var/www/html:
    `cd /var/www/html`
    Now fetch latest source and checkout:
    ```
    sudo -u www-data -s
    git fetch origin
    git checkout origin/gphoto-preview
    ```
    Update submodules and build the css files:
    ```
    git submodule update --init
    yarn install && yarn build
    ```
    Once that's done you're ready
    `exit`

2. If you have installed Photobooth via git from a different fork:
    Open a Terminal and navigate into your installation folder, by default /var/www/html:
    ```
    cd /var/www/html
    sudo -u www-data -s
    ```

    Add my repo as remote:
   `git remote add andi34 https://github.com/andi34/photobooth`

    Now fetch latest source and checkout:
    ```
    git fetch andi34
    git checkout andi34/feature/gphoto-preview
    ```
    Update submodules and build the css files:
    ```
    git submodule update --init
    yarn install && yarn build
    ```
    Once that's done you're ready
    `exit`

** *Please leave some feedback* **

Acemond, 2020-07-25:
Wow how did you get to see that code? Did you check the forks of the project? I've litterally spent hours pulling my hairs on that feature on a fork of this repo :D

Didn't manage to make it work in my case because, I believe, some libraries were missing on my raspberry. What did you add to make it work?

Acemond, 2020-07-25:
My code was mostly experimental stuff to get something to work, quite not huh... best practices friendly :o

andi34, 2020-07-25:
@Acemond yeah, from time to time I am looking around on new forks. ;)

Looked like something to work with, can't test myself currently so I don't know if it works now.
Waiting for feedback 👍

Acemond, 2020-07-25:
Well it suffered two main issues, first it would only display the first image from the preview video (but that might be the library issue from my side), but more importantly there was an issue between the preview and the photo where if the length on the video overlapped the moment where it would ask for the photo it crashed because it cannot gain access to USB still in use. The problem is that actual length of the video in seconds depended on the raspberry performances.

andi34, 2020-07-25:
Thanks for the information! There's some parts which could be handled different, not sure if this could help with the performance.

#### Another way
> Hi,
> After multiples test I got this working:
>     (...)
> For it to work on startup I had to edit /etc/rc.local:
> `modprobe v4l2loopback exclusive_caps=1 card_label="GPhoto2 Webcam"`
> `rmmod bcm2835-isp`
> 
> rmmod bcm2835-isp was needed as chromium was taking this device instead of the V4l2 one.
> 
> **How it work**
> We use v4l2loopback to create a virtual device and `gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0` to send the photo preview to it.
> On "take pic" button we start the process and end to before the picture is taken to free gphoto2.
> 
> It is a ugly draft, I hope I will improve it and post the update here.

Comment2
> I pushed the draft here https://github.com/couz74/photobooth.
> There is still two issue:
> 
> * Sometime I need to run `sudo gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0` first and load chrome a first time with a webpage asking for the camera. If I don't do so sometime chrome doesn't detect the V4l2 camera launch from php.
> * On the takeVideo.php I need to use a sleep() method in order to allow the stream to launch, still need to figure a way to have something cleaner

Comment 3
> Hi,
> In the admin panel the setting "See preview by device cam" must indeed be set.
> Then there are three things:
> 
> * `modprobe v4l2loopback exclusive_caps=1 card_label="GPhoto2 Webcam"`  create the virtual webcam and `sudo gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0` in the takeVideo.php send the outpout of the Gphoto command to the /dev/video0 device (you can use v42l-ctl --list-devices to check which /dev/* is the correct one).
> * The second thing is the issue with Chromium (did not test firefox), if there is another webcam like
>   `bcm2835-isp` in my side it take it by default instead of the Gphoto one... therefore I always do `rmmod bcm2835-isp`.
> * Finaly Chromium sometimes also doesn't detect the Gphoto webcam if it wasn't loaded  first with a Gphoto Webcam having the Ffmjpg feed . To do that I send the command in a shell  `sudo gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0`  then I start Chromium and check if a testing webcam website work.
> 
> As you can see there is still some issue with the solution XD

https://github.com/couz74/photobooth/commit/4b5ba4c797571e45214b3cc040fe229ed605ba89

rover1982, 2020-08-04:
Hi ;) 
I'm not a pro with this git and raspberry thing- so please excuse eventually stupid questions.
It's quite different to c and embedded system programming ;)

So- you photobooth is great. There's just a few problems i have:
-Using an EOS60 D/ Taking pictures works fine
-Remote preview does not work with chromium ( camera doesn't get command, the canon normally makes a click when starting 
  the live view mode)
-Tried the same in firefox- now it works better, but no live picture in countdown screen. 
  ( I know camera works- there is a click when starting the countdown, click when cheese, and then the photo.
  You just have to be shure that if you change the countdown you also have to change the video duration ;)

In admin panel 'use remote preview' is set, 'preview by device cam' can't be activated (as mentioned above). I switch it on- after the next actualisation it is off again.

What am i doing wrong ?

andi34, 2020-08-04:
Preview by device cam only used the device cam, e.g. tablet camera, smartphone camera or raspi cam.
So if we like to use the preview from gphoto we have to disable all other preview options to prevent issues. This is done on save.

This is only WIP and needs to be adjusted in future, it's not even tested by myself as I don't have a camera with that feature atm.

It's atm up to the community to improve it further.

strunki, 2020-09-25:
Hey andi34,

first of all thank you for constantly working on the possibility to make the preview happen!
I used your experimental feature to preview from a Nikon D5100 on Raspberry Pi 4B+ 4GB. It's actually starting the video and ending it after 10 seconds to allow the photograph. 
When I searched your changes I first found the hard coded 10 seconds of --capture-movie that is not mentioned in any documentation in here, but leads to an error sent by the Photobooth webpage if the Timer is set <10. I just wanted to mention that if somebody is running into the same issue like I were. 
So far so good, after changing this i ran into the next issue that I wasn't able to display the mjpg file in the countdown screen. I still didn't find out how to get this right. Might this be a consequence of my setup, accessing the Pi via it's IP from the browser on a different machine in the LAN? I thaught about the problem of accessing the photobooth via port 80 but cannot use 8090 to get out the image, or am I running in a completely wrong direction?

Would be happy to hear from you
strunki

andi34, 2020-09-25:
Yeah, it's WIP and not my implementation. Found it on GitHub and made it in a nice shape to work on it further.

You should access Photobooth from your Pi and open via "localhost".

Not tested, but maybe you can replace the `localhost` inside the config. (further changes needed to access the stream via IP address?) 

Command to trigger the remote preview:  
`gphoto2 --capture-movie=10s --stdout | ffmpeg -re -i pipe:0 -listen 1 -s 960x640 -f mjpeg http://YOUR-IP:8090/video-stream.mjpg'`

and

Remote-Preview-URL:  
`http://YOUR-IP:8090/video-stream.mjpg`

closed 2020-11-17 by andi34