andi34/photobooth #87: white banner crashes userinterface

ChrisNi91, 2020-07-30:
Hey Andi,
I have a white banner crashing the userinterfaces since a couple of days. This appeared without any error messages or changes from me side.

I tried severel things via adminpanel but nothing really worked. What I did so far:

1) Changing the color, theme and background pictures
2) activating and deactivating of any kind of panels (e.g. fork banner)
3) I deleted the whole CSS Folder und reupdated to the newes version

The banner is still there:
[image]

Do you have any idea how to solve that?
many thanks!

andi34, 2020-07-30:
You're using v2.7.2?

ChrisNi91, 2020-07-30:
Jap, I do.

andi34, 2020-07-30:
Could you please see if zooming fixes it?
Press Ctrl and scroll on your mousewheele to zoom in, maybe the background doesn't match your screen resolution?

If that doesn't work, guess you're using /var/www/html as root of Photobooth, open a terminal and type the following:
```
sudo chown -R www-data:www-data /var/www/
sudo -u www-data -s
cd /var/www/html
git commit -am "my changes"
git checkout -b backup-20200730
git remote add andi34 https://github.com/andi34/photobooth
git fetch andi34
git checkout andi34/stable2
git submodule update --init
yarn install
yarn build
mv config/my.config.inc.php config/my.config.inc.php.bak
exit
```
Also, if exist
`sudo rm /var/www/html/admin/config.json`

andi34, 2020-07-30:
And if you're interested: 
https://t.me/PhotoboothGroup

andi34, 2020-07-30:
And another thing: try from a private-/incognitotab or delete your browsers cache - sometimes the browser cache can cause some trouble after changes.

ChrisNi91, 2020-07-31:
Hey Andi,

thank you very much for your fast reply.

It seems that your last suggestion works out. By using an incognito tab or deleting the cache the benner dissapear.
I'm still a bit confused that it happend suddendly and simulatinously on both devices (Laptop and Android tablet). But I guess I'm fine now.

Do you see any solution to fix that software side?

andi34, 2020-07-31:
Great to hear it works now.
Atm there's no solution known to me which could be implemented and would declare it as browser issue. But I'll keep my ears and eyes open if there's anything we can do Photobooth side.

commit 2b284266 (2019-10-07): fix webcam preview

fix #87
closed 2020-07-31 by andi34