andi34/photobooth #95: Externer Blitz über raspberry ansteuern

ozke86, 2020-08-13:
Hi 
Erst Mal tausend Dank für dein Projekt !

Ich habe leider eine Lichtschwäche Nikon p900
An der ich keinen externen Blitz anschließen kann 
Jedoch hab ich einen externen Blitz/Fotoleuchte von neewe!
Mit sync (Klinke Anschluss)

Ist es möglich ein Signal vom Klinke Anschluss des pi's zu schicken das der neewer Blitz verarbeiten kann ?
Natürlich übers das Webinterface bearbeitbar ?

Wir haben am 26.9 Hochzeit ;)

andi34, 2020-08-13:
Hey!

Tut mir leid, da ist mir kein Weg bekannt zur Zeit.
Ich lasse das als Feature request gerne offen, falls wer anderes Lust und Zeit hat sich das anzuschauen. Ich habe noch ein paar andere Features in Planung um welche ich mich erst einmal kümmern werde.

Es wäre nur schön wenn das Template für die Feature request benutzt wird:
https://github.com/andi34/photobooth/blob/dev/.github/ISSUE_TEMPLATE/feature_request.md
(Einfach den Eingangspost entsprechend editieren)

Beste Grüße
Andi

Nie-Oh, 2020-08-21:
Hallo,

also eine der Blitzsteuerung über den Raspi stelle ich mir schwer vor. Der Blitz muss ja auch mit der Kamera Syncron sein usw.
Das einfachste wäre, den Blitz über eine Fotozelle auszulösen. Schau mal bei e...y oder a....on nach Servo Blitzauslöser. Das ist am einfachsten und schnell umgesetzt. Eventuell hat den Blitz schon solch eine Funktion.

Gruß
Steffen

closed 2021-01-02 by stale[bot]