andi34/photobooth #110: Polish translation

badyl00, 2020-09-16:
**Missing language**
Polish

**Additional context**

andi34, 2020-09-16:
https://crowdin.com/project/photobooth/pl#

Please join as Translator.

Thanks for your request ☺️

andi34, 2020-09-16:
Will be added once translation is at least 70% completed.
If you have questions please let me know. Thanks a lot for your contribution!

badyl00, 2020-09-21:
I added translations. All of it its made on computer and must check it on
working hardware (raspberry). It is possibile to emulate aplication on
computer (windows) to check translations and functions ? Thanks.

andi34, 2020-09-21:
Hey!
Thanks a lot! Will take a look at it later this week.
I can send you a zip with it included if you like. On windows there might be an issue because of simple-translator having an `@` in the path, need to investigate. In general you can also setup an apache Webserver with php enabled.

andi34, 2020-09-27:
[image]

[image]

[image]

closed 2020-09-27 by andi34