andi34/photobooth #113: Syncing photos to external drive (also backup)

thatonedude3470, 2020-09-24:
**Is your feature request related to a problem? Please describe.**
Currently there is no default way of backing up photos from the photobooth to an external drive (e.g. USB or HDD) despite manually coping them, 

**Describe the solution you'd like**
My idea was to have a drop down in the admin to select a drive, then if a drive is selected a cronjob could rsync the files onto it periodically.

**Describe alternatives you've considered**
For Linux there are udev that could be triggered once a mountable drive was inserted, but that doesn't port to Windows (maybe there are ways for Windows to achieve the same).

andi34, 2020-09-26:
Not sure if there's a good way to do it without security impacts. I'd not like to give to much permissions to the www-data user.
What about running e.g. [freefilesync](https://freefilesync.org/) in background?
Or setting up a cron job on Linux to copy files from location a to location b?

I've been working on a solution to download the data folder as zip from disc-usage page , but not got it working last time I've forward ported it to latest version (need to look at it again as it worked when I started working on it).

thatonedude3470, 2020-09-26:
Not sure how much permissions lsblk needs, but the selection could be realized with it (newer version can also export to JSON see [systeminformation](https://github.com/sebhildebrandt/systeminformation/blob/1dd402812396d35b6007b497263dc712f1c347ba/lib/filesystem.js#L754)).
The selected drive name could then be written to the/a config file and then read from a script that gets executed via a cronjob. (cronjob triggers script every 5 mins -> script reads which drive to use -> mount drive if not mounted yet -> copy files). 
In that manner the script won't be executed from the www-data user. 

Could the zip solution also be offered to visitors? You got me interested there now.

andi34, 2020-09-27:
> Not sure how much permissions lsblk needs, but the selection could be realized with it (newer version can also export to JSON see [...]

Currently I've only limited time to work on the Photobooth and there's some stuff left on my to-do list first.
Maybe you've time to work on a solution?
Else I'll take a look at it at some point later.

> Could the zip solution also be offered to visitors? You got me interested there now.

Once there's something ready to test yes :)

thatonedude3470, 2020-09-27:
Sure, understandable. I'll take a spin at it and open a PR once it's stable :)

> Once there's something ready to test yes :)

Really looking forward to it! Guess it will come with some password option?

thatonedude3470, 2020-09-30:
Tested on a fresh Raspberry PI install and it worked without any issues. Synced all the files in data folder to a USB stick.
Only issue I could see for now is that the cronjob gets added to root and thus the script called by root, due to the install script being executed as root.
The script works without root privileges, so maybe we could/should redirect the crontab call to another user?
But for now that should do it :)

andi34, 2020-10-31:
> Could the zip solution also be offered to visitors? You got me interested there now.

https://github.com/andi34/photobooth/pull/125

PR #158 (2020-12-23): Feature: add filesync
#### Prerequisites checklist

- [x] I have read the [contributing guidelines](https://github.com/andi34/photobooth/blob/dev/CONTRIBUTING.md).

#### What is the purpose of this pull request? (put an "x" next to an item)

- [x] New feature, implemented by @thatonedude3470 and @jacques42

#### What changes did you make? (Give an overview)

- Added rsync to common packages in install-raspbian.sh
- Added cronjob via install-raspbian.sh (needs documentation in Wiki for manual install)
- Added new sync-to-drive.js which does all the logic for selecting and starting rsync
- Added [... cut]
closed 2020-12-23 by andi34