andi34/photobooth #139: vertical scrolling neccessary on Galaxy Tab S2 in Chroma-Key-Editing

danielh84, 2020-11-21:
**Describe the bug**
With the Galaxy Tab S2 (1024x762, Device-Pixel-Ratio:2) you get an vertical scrollbar by editing pictures with chroma key (chromakeying.php?filename=20201121_125652.jpg). Maybe the main picture should be smaller to get everything on the tablet without scrolling. You cant see the "save" and "close" buttons without scrolling:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Use Ratio of the Galaxy Tab S2
2. Go to "Gallery" and use "Chroma Key"-Button
3. See error

**Expected behavior**
You have to see the save and close Buttons without scrolling. 

**Environment (please complete the following information):**
 - Android
 - Fully Kiosk Browser

**Photobooth**
- Which version of Photobooth are you using? -> Develop-branch 21.11.2020

andi34, 2020-12-03:
Hey and sorry for the late reply.

You can try to lower the width here:
https://github.com/andi34/photobooth/blob/3c6a7bcdc5897c727abfb06d3fc1e4afa3f67ae3/src/sass/chromakeying.scss#L24

danielh84, 2020-12-07:
width: 850px; works fine with this device. 

i will try to find a css-config that it will fit to all resolutions. maybe there is a good opinion.

andi34, 2020-12-07:
Something like

```
@media (max-height: 765px) {
  .canvasWrapper {
    width: 850px;
  }
}
```
Maybe?

andi34, 2020-12-19:
@danielh84 can you try from `feature/livekeying` branch?

danielh84, 2020-12-21:
no, does not work. Tested with classic and modern styling

andi34, 2021-01-28:
After https://github.com/andi34/photobooth/pull/159 you're allowed to use your cust style

Needs all of them to work (can be copy of existing files):
```
template/custom.template.php
resources/css/custom_style.css
resources/css/custom_chromakeying.css
resources/css/custom_live_chromakeying.css
```

andi34, 2021-02-28:
Any progress on the css changes to not affect all resolutions?

commit 54fd4eb1 (2019-10-19): fix deletion of db file

fix #139
closed 2021-03-25 by andi34