andi34/photobooth #156: Enhance collage display

GeoffreyV, 2020-12-15:
**Is your feature request related to a problem? Please describe.**
When using collage and standalone picture, the collage display is 90° rotated 

**Describe the solution you'd like**
- [ ] Rotate collage when displaying it if necessary (2x4 and 2x4BI for example)

- [ ] Not display twice the pictures for collage where some photos are duplicated (2x4 for instance) 
_The collage is optimized for printing but not for display (small screen)_

**Describe alternatives you've considered**

When using collage only (see #155), setting 90 to preview rotation is a workaround

andi34, 2020-12-30:
Thanks for your suggestion, I'll try to find a solution

andi34, 2020-12-30:
Please take a look at https://github.com/andi34/photobooth/pull/163, it'll rotate the picture if needed and also rotate the final collage if needed.

andi34, 2021-01-28:
Fixed by https://github.com/andi34/photobooth/pull/163,
On 2x4 displaying the same 4 images twice is what we want (not changed).

closed 2021-01-28 by andi34