andi34/photobooth #164: More Hardwarebuttons for Photobooth

Hubert-73, 2020-12-31:
I would like to make the following suggestion:

Please integrate the option to select multiple buttons.  For example: single photo, collage, print, etc.

lasolco, 2021-01-03:
I have a same idea. I will create a script...

jacques42, 2021-01-03:
@lasolco would you mind to check how to make this enhancement in the `src/js/remote_buzzer.js` script ? @Hubert-73 currently is using this script and it should be fairly straight forward to add monitoring for two more RPIO pins. The collage trigger via a separate PIN should be simple to add in, along with a logic to disable long-press for collage if a second button is connected. Notificaion via io messaging is already implemented.

A third PIN monitoring for Print should also be feasible, yet this will require to enhance `core.js` as to trigger print currently is not support via the io client / server flow. 

I am currently working a pretty comprehensive re-write of the admin panel, configuration files and translations of panel content and manual, hence lack time to work on this one already now. But once that admin panel work is done I am more than happy to engage, I do like the feature idea a lot.

jacques42, 2021-01-23:
Bump. More people are asking for additional HW buttons, I.e to allow a clean Raspi shutdown.

lasolco, 2021-01-23:
Hi,
I suggest the following style. I'm sorry, I didn't integrated it in your code. I just created a new Python File, which is always started in runtime. In my opinion, it is much more easy and flexible. So you can also create shortcuts or buttons for shutdown and everything else.

-------------------------------------------
# -*- coding: utf-8 -*-
from RPi import GPIO
from datetime import datetime
import os, time, uinput
 
GPIO.setwarnings(False) # Ignore warning for now
GPIO.setmode(GPIO.BCM) 
 
GPIOPinF = 16
GPIOPinD = 12
 
GPIO.setup(GPIOPinD, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(GPIOPinF, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
 
try:
    while True:
        time.sleep(0.05)
        input_stateF = GPIO.input(GPIOPinF)
        if input_stateF == True:
            with uinput.Device([uinput.KEY_PAGEUP]) as device:
             time.sleep(0.5)
             device.emit_combo([uinput.KEY_PAGEUP])
             time.sleep(2)
        input_stateD = GPIO.input(GPIOPinD)
        if input_stateD == True:
            with uinput.Device([uinput.KEY_PAGEDOWN]) as device:
             time.sleep(0.5)
             device.emit_combo([uinput.KEY_PAGEDOWN])
             time.sleep(2)
except:
    GPIO.cleanup()
-------------------------------------------

#pyhton script to run on every startup
crontab -e
#paste entry at the end
@reboot python /home/pi/mypfad.py &

Beste Grüße

lasolco, 2021-01-23:
Danke für das Editieren. Ich war gerade selbst dabei :D

jacques42, 2021-02-17:
This feature will start become available in PB3 as part of Remote Buzzer.

jacques42, 2021-03-07:
v3 now implements separate buttons for picture, collage and shutdown. Also v3 implements a rotary encoder control - this one is extremly nice, one can operate PB with literally one knob / button.

Next buttons will be "print" and "QR code".

jacques42, 2021-03-08:
Hardware button support for PRINT added

jacques42, 2021-03-31:
PR https://github.com/andi34/photobooth/pull/221 adds Rotary support for Delete button. Please read the notes for required admin panel settings

closed 2021-04-03 by andi34