andi34/photobooth #166: Collage - retake a single picture

lasolco, 2021-01-03:
Hi, I noticed during a collage a nice new feature.

For example, you do 4 pictures during the collage, but during the 2nd picture a mistake took place. So it would be great to have the possibility to retake the picture or to request the user to accept a picture for the collage.

I would like to suggest to add the possibility to request the user after each picture of the collage to acknowledge the picture or do a retake.

Thank you very much!

Best regards

andi34, 2021-01-03:
Hey @lasolco !
Thanks for your suggestion. I am about to rework some code of the collage creation and will take a look on your idea once I've more time (there's some other stuff on my to-do-list too which I need to finish first).

For future feature request(s) please use the feature request template.

Best regards

Andi

andi34, 2021-02-07:
Started working on it.

[image]

Todo: 
- don't repeate picture
- delete old picture to take a new one

andi34, 2021-02-07:
Will be included in next release

closed 2021-02-07 by andi34

jacques42, 2021-02-07:
Did you get to test if this also works with an active RemoteBuzzer integration ? Let me know in case not and I'll have a look

andi34, 2021-02-07:
No, would be great if you're able to test. But I have to push one fix later.

andi34, 2021-02-07:
[image]