andi34/photobooth #182: Rotary Encoder für die Website Bedienung

DIY89, 2021-02-16:
Hi Andi,

da meine Photobox kein Touchscreen enthält würde die Website gerne mit einem Druck-Dreh-Taster bedienen wollen.
In der vorherigen Version habe ich ein Sock-Server(nodejs) eingebaut der hardwareseitig die Signale einliest und an den Client (Website) Befehle (links/rechts/gedrückt) sendet.

Die Befehle habe ich genutzt um von einem zum anderen benachtbarten Button zu springen. 
Beim Aufruf der Startseite wird dem ersten Button eine zusätzlich css-class spendiert "focused".
Über den Javascript Code springe ich je nach Befehl zum $('.focused').prev('button') oder zum $('.focused').next('button')

Aktuell funktioniert die Verbindung zu dem sock-Server nicht da der Cross-Origin Zugriff nicht funktioniert.
Die Webserver Adresse steht auf Localhost (127.0.0.1). Selbt wenn ich die Website auf dem PI aufrufe klappt die Verbindung zu dem Sock-Server nicht.

`/* VARIABLES */
const myPid = process.pid;
let mTimeTrigger = 0,
    collageInProgress = false;

let triggerArmed = true,
    buttonIsPressed = false;

const Rotary = require('raspberrypi-rotary-encoder');

// WARNING ! This is WIRINGPI pin numerotation !! please see https://fr.pinout.xyz/pinout/wiringpi#*
const rotary = new Rotary(12, 13, 14);

/* HANDLE EXCEPTIONS */

process.on('uncaughtException', function (err) {
    console.log('socket.io server [', myPid, ']: Error: ', err.message);
    fs.unlink(pidFilename, function (error) {
        if (error) {
            console.log('socket.io server [', myPid, ']: Error deleting PID file ', error.message);
        }
    });
    console.log('socket.io server [', myPid, ']: Exiting');

    /* got to exit now and here - can not recover from error */
    process.exit();
});

/* SOURCE PHOTOBOOTH CONFIG */

const {execSync} = require('child_process');
const stdout = execSync('cd api && php ./config.php').toString();
//const config = JSON.parse(stdout.slice(stdout.indexOf('{'), -1));

/* WRITE PROCESS PID FILE */

const pidFilename = config.folders.tmp + '/socket_server.pid';

const fs = require('fs');

fs.writeFile(pidFilename, myPid, function (err) {
    if (err) {
        throw new Error('Unable to write PID file [' + pidFilename + '] - ' + err.message);
    }

    console.log('socket.io server [', myPid, ']: PID file created [', pidFilename, ']');
});

/* START WEBSOCKET SERVER */

console.log(
    'socket.io server [',
    myPid,
    ']: Requested to start on http://' + config.webserver_ip + ':' + config.socket_port,
    ', Pin_l: ', config.pin_l,
    ', Pin_r: ', config.pin_r,
    ', Pin_b: ', config.pin_b
);

rotary.on("rotate", (delta) => {
  console.log("Rotation :"+delta);
  if(delta > 0){
     ioServer.emit("photobooth-socket", "negative");
  }else{
     ioServer.emit("photobooth-socket", "positive");
  }
});
rotary.on("pressed", () => {
  console.log("Rotary switch pressed");
  ioServer.emit("photobooth-socket", "pressed");
});
rotary.on("released", () => {
  console.log("Rotary switch released");
  ioServer.emit("photobooth-socket", "released");
});

const ioServer = require('socket.io')(config.socket_port, {
    cors: {
        origin: 'http://' + config.webserver_ip,
        methods: ['GET', 'POST']
    }
});

ioServer.on('connection', function (client) {
	console.log('socket.io server [', myPid, ']: New client connected - ID', client.id);
	client.on('photobooth-socket', function (data) {
		console.log(
			'socket.io server [',
			myPid,
			']: Data from client ID ',
			client.id,
			': [ photobooth-socket ] =>  [',
			data,
			']'
		);

		/* Main */
		if($('#start').is(':visible')){
			// set focus if not done yet
			if(!($('.divinnen2 a').hasclass('focused'))){
				$('.divinnen2 a').first().addClass('focused');
			}
		}

		/* Result */
		if($('.resultInner').hasClass('show')){
			// set focus if not done yet
			if(!($('.resultInner a').hasclass('focused'))){
				$('.divinnen2 a').first().addClass('focused');
			}
		}

		/* Filter */
		if($('#mysidenav').hasClass('sidenav--open')){
			// set focus if not done yet
			if(!($('#mysidenav div').hasclass('focused'))){
				$('.divinnen2 div').first().addClass('focused');
			}

			if(data == "negative"){
				if($('.focused div:first').hasClass('focused')){
					$('.focused').removeClass('focused').prev().addClass('focused');
					return;
				}
			}
		}

		/* Gallery */
		if($('#gallery').hasClass('gallery--open')){
			// set focus if not done yet
			if(!($('gallery__body div').hasclass('focused')) || (!($('gallery__close').hasclass('focused')))){
				$('.gallery__body a').first().addClass('focused');
			}

			// Up
			if(data == "negative"){
				// if focus is on picture
				if($('.pswp__button.focused').nextAll('.pswp__button:visible:first').length != 0){
					$('.focused').removeClass('focused').nextAll(':visible:first').addClass('focused');
				// if focus is on close button
				}else if($('gallery__close').hasClass('focused')){
					$('.focused').removeClass('focused');
					$('.gallery__body a').first().addClass('focused');
				}
			}

			if(data == "positive"){
				// if focus is on picture
				if($('.pswp__button.focused').prevAll('.pswp__button:visible:first').length != 0){
					$('.focused').removeClass('focused').prevAll(':visible:first').addClass('focused');
				}else if($('gallery__close').hasClass('focused')){
					$('.focused').removeClass('focused');
					$('gallery__close').addClass('focused');
				}
			}
			console.log('Focuse now on: ' + $('.focused').attr('class') + ' FocusIndex: ' + $('.focused').index());
			return;
		}

		/* Photoswipe */
		if($('.pswp').hasClass('pswp--open')){
			if(data == "negative" && $('.pswp__button.focused').nextAll('.pswp__button:visible:first').length != 0){
				$('.focused').removeClass('focused').nextAll(':visible:first').addClass('focused');
			}
			if(data == "positive" && $('.pswp__button.focused').prevAll('.pswp__button:visible:first').length != 0){
				$('.focused').removeClass('focused').prevAll(':visible:first').addClass('focused');
			}
		}

		/* set focus to prev() or next() button */
		if(data == "negative"){
			$('a .focused').prev().removeClass('focused').next().addClass('focused');
		}
		if(data == "positive"){
			$('a .focused').next().removeClass('focused').prev().addClass('focused');
		}
		if(data == "pressed"){
		$('.focused').removeClsss('focused').click();
	}
});

	client.on('disconnect', function () {
		console.log('socket.io server [', myPid, ']: Client disconnected - ID ', client.id);

		if (ioServer.engine.clientsCount == 0) {
			console.log('socket.io server [', myPid, ']: No more clients connected - removing lock and arming trigger');
			triggerArmed = true;
			collageInProgress = false;
		}
	});
});
console.log('socket.io server [', myPid, ']: socket.io server started');

`

jacques42, 2021-02-16:
Hi @DIY89 - thank you for the idea and I like it a lot. I think we should add it in as base functionality. And this idea could also work in a scenario with one, two or three distinct buttons -  i.e. one button for left, another one for right and one more for press. Or in a a scenario with one button where short press moves focus "to the right" - this way one can cycle through UI - and a longer press issues a button press. So really great idea and could help to build very simplistic, button based UI interaction. 

Instead of building another, separate server script, I'd integrate the functionality as option to the existing remote-buzzer server. From my perspective it seems fairly possible to integrate your idea into that existing script. (there is rework going on behind the scenes which is not yet shared publicly). 

I need to check into why CORS prevents localhost to localhost  - I never tested this. 

Checking the `raspberrypi-rotary-encoder` library, I would propose that we find an alternative or do a native encoding. Main reason here is that it's not a native JS library but the JS is a wrapper for a python script being forked - that is something we should be able to avoid having to do. And the python library then uses the `wiringpi` library for GPIO access, where I just converted remote-buzzer to using the `onoff` library. When you started looking around, was there alternative libraries you could see ? 

So - I'm happy to take the remote-buzzer server part if that would look like help, and then with your help to complete the JS script updates for UI navigation, we can make this a quite nice feature. What do you think ?

jacques42, 2021-02-17:
Hi @DIY89 wenn du willst komm in unseren Telegram Channel mich würde auch interessieren welchen Rotary Button du verwendest

DIY89, 2021-02-17:
> Hi @DIY89 wenn du willst komm in unseren Telegram Channel mich würde auch interessieren welchen Rotary Button du verwendest 

Ihr könnt mich gerne in den Channel aufnehmen.

DIY89, 2021-02-17:
> > Hi @DIY89 wenn du willst komm in unseren Telegram Channel mich würde auch interessieren welchen Rotary Button du verwendest 
> 
> Ihr könnt mich gerne in den Channel aufnehmen.

Den Druckdrehtaster habe ich verwendet: https://www.reichelt.de/drehimpulsegeber-15-impulse-30-rastungen-vertikal-stec11b04-p73914.html?&nbc=1

DIY89, 2021-02-19:
> Hi @DIY89 - thank you for the idea and I like it a lot. I think we should add it in as base functionality. And this idea could also work in a scenario with one, two or three distinct buttons -  i.e. [...]

I had tried to integrate the sock server in the existing 'remote-buzzer-server' without success like I wrote before. I would prefer to implement an option for the art of remote-control (push-buttons, rotary-encoder, touch). To avoid the detour via python the following libary could be used (https://www.google.com/url?sa=t&source=web&rct=j&url=https://github.com/cruepprich/node_rotary&ved=2ahUKEwiEgrXlm_buAhUGDxQKHTMEClQQFjACegQICBAD&usg=AOvVaw0j-yMpGsYt3B5nAaaMxREk&cshid=1613747605233) - not tested yet.

jacques42, 2021-03-02:
@DIY89 I pushed branch [v3/hardwarebutton-rotary](https://github.com/andi34/photobooth/tree/v3/hardwarebutton-rotary) which has a well working base version of a rotary integration. Have a look please and I'd appreciate if you could do some testing. Please also check the respective [FAQ section](https://github.com/andi34/photobooth/blob/v3/hardwarebutton-rotary/faq/faq.md#can-i-use-hardware-button-on-my-raspberry-pi-to-take-a-picture-) for functional details. Any feedback and suggestions is really appreciated.

DIY89, 2021-03-16:
Hallo,

it's great to see what you're doing for the rotary-encoder function.
But i tried to get the feature running on my raspi. I still struggling with cors exception.

The socket-server seems to run when i check the logs.
`pi@raspberrypi:/var/www/photobooth/data/tmp $ cat remotebuzzer_server.log`
`Remote Buzzer Server [2859]: Server starting on http://127.0.0.1:14711`
`Remote Buzzer Server [2859]: socket.io server started`
`Remote Buzzer Server [2859]: Operating in ROTARY mode`
`Remote Buzzer Server [2859]: Ready for Rotary Encoder on GPIOs 21 20 16`
`Remote Buzzer Server [2859]: Initialization completed`
`Remote Buzzer Server [2859]: PID file created [ data/tmp/remotebuzzer_server.pid ]`

Console
`Remote Buzzer client is enabled remotebuzzer_client.js:11`
`Remote buzzer connecting to http://127.0.0.1:14711 remotebuzzer_client.js:36 `

`Access to XMLHttpRequest at 'http://127.0.0.1:14711/socket.io/?EIO=4&transport=polling&t=NWyA3kU' from origin 'http://localhost' has been blocked by CORS policy: The 'Access-Control-Allow-Origin' header has a value 'http://127.0.0.1' that is not equal to the supplied origin.`
`GET http://127.0.0.1:14711/socket.io/?EIO=4&transport=polling&t=NWyA3kU net::ERR_FAILED`

Any idea what to do to fix this issue?

andi34, 2021-03-16:
@DIY89 can you add the real IP address inside the Adminpanel and access it though the real IP?

Instructions:
https://github.com/andi34/photobooth/blob/21070854fecbb70bc3377ecf9891681f1e21119d/faq/faq.md#can-i-use-hardware-button-on-my-raspberry-pi-to-take-a-picture-

DIY89, 2021-03-16:
I'd tried several IPs:
`$config['webserver']['ip'] = '127.0.0.1';` --> Access via Raspi Desktop --> Chrome Browser
`$config['webserver']['ip'] = 'localhost';` --> Access via Raspi Desktop --> Chrome Browser
`$config['webserver']['ip'] = '192.168.1.46';` --> Access from my PC (192.168.1.20) --> Chrome Browser

Nothing is working

andi34, 2021-03-16:
@jacques42 you've modified your server config?
https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSMissingAllowOrigin

I've not had time yet to test myself.... 😅

DIY89, 2021-03-16:
The settings in the `remotebuzzer_server.js` seems to be correct.

`const ioServer = require('socket.io')(config.remotebuzzer.port, {
    cors: {
        origin: 'http://' + config.webserver.ip,
         methods: ['GET', 'POST']
    }
});`

[https://socket.io/docs/v3/handling-cors/](https://socket.io/docs/v3/handling-cors/)

jacques42, 2021-03-16:
I can't test this, in my setup it works like a charm. I do use Firefox or Chrome on client-side. 

I understand the error message as the CORS permission is for say 127.0.0.1 but the request origins from server localhost. That does not match and the error occurs. But I do not know on why the origin server name is localhost in the header this is weird.

Can you try to modify 'src/js/remotebuzzer-server.ja' to permit all origins. Example:

'''

const io = require('socket.io')(server, {
  cors: {
    origin: '*',
  }
});
'''

Once you changed the file you need to run 'yarn build' and make sure the server process restarts (save admin Config / reload main page)

Would that make a difference ?

jacques42, 2021-03-17:
I changed the cors setting to permit any origin and pushed it to the branch (commit https://github.com/andi34/photobooth/pull/202/commits/2e7f76d31bbc88b02a29343e001a28a78d8c05d5) 

Still works like a charm on my setup. Note that if the server is not yet fully started, the client might throw one or two CORS error message in the console log. As soon as the server is fully started up, the errors stop and the feature starts working.

Please try again with the latest updates and let me know if this makes a difference for you.

DIY89, 2021-03-17:
I tried all your suggestions, but still have the problem.

Do you have an .htaccess file in your project folder with cors entries or do you have some special entries inside of the webserver config `php.ini`.

DIY89, 2021-03-18:
I just tried a small socket server example on my local machine with xampp [https://github.com/DIY89/socket-connection.git](https://github.com/DIY89/socket-connection.git).

No cors exception occured.
Afterwards i switched to my Raspi and receive the cors exception again.

I think there is something in the Webserver Config.

I'll report.

jacques42, 2021-03-18:
Thank you for investigating. Please consider that the socket.io webserver is established by `nodejs` - the socket.io websockets are not served by Apache or nginx. You can check this by running 
```
root@raspberrypi:/var/www/html/andi34# netstat -antp |grep 14711
tcp6       0      0 :::14711                :::*                    LISTEN      6063/node
tcp6       0      0 172.16.35.254:14711     172.16.35.59:56714      ESTABLISHED 6063/node
```

I'd be interested to see the HTTP protocol exchange on the socket.io connection when the error occurs. Would you be able to do the following commands, in a setup where the server ip in Photobooth is configured as `192.168.1.46` and you access the PB homepage from your PC at `192.168.1.20`. This is needed so that `tcpdump` can actually see the packets. 

1. Close browser, start Photobooth on the Raspi, including `remotebuzzer_server`. Check the log to ensure it's running
2. Start the following command on the Raspi - it will dump the HTTP protocol exchange to a file
```
/usr/sbin/tcpdump -A -s 10240 "tcp port 14711 and (((ip[2:2] - ((ip[0]&0xf)<<2)) - ((tcp[12]&0xf0)>>2)) != 0)" | egrep -a --line-buffered ".+(GET |HTTP\/|POST )|^[A-Za-z0-9-]+: " | perl -nle 'BEGIN{$|=1} { s/.*?(GET |HTTP\/[0-9.]* |POST )/\n$1/g; print }' > /tmp/pb_socketio_tcpdump.txt
```
3. Open the browser on your PC and load the Photobooth homepage. Maybe reload it two times or so
4. Kill the command and share the content of `/tmp/pb_socketio_tcpdump.txt' 

For reference, a successful connection on my end looks like this, where 172.16.35.254 is the server. Note that the CORS setting is set to "permit any (*)" as specified by the server in the `Access-Control-Allow-Origin` header. 
```
GET /socket.io/?EIO=4&transport=polling&t=NX79GVI HTTP/1.1
Host: 172.16.35.254:14711
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:86.0) Gecko/20100101 Firefox/86.0
Accept: */*
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Origin: http://172.16.35.254
DNT: 1
Connection: keep-alive
Referer: http://172.16.35.254/andi34/

HTTP/1.1 200 OK
Access-Control-Allow-Origin: *
Content-Type: text/plain; charset=UTF-8
Content-Length: 96
Date: Thu, 18 Mar 2021 21:01:52 GMT
Connection: keep-alive

POST /socket.io/?EIO=4&transport=polling&t=NX79GbW&sid=cy3kaa8c_wHcgU-CAAAG HTTP/1.1
Host: 172.16.35.254:14711
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:86.0) Gecko/20100101 Firefox/86.0
Accept: */*
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Content-type: text/plain;charset=UTF-8
Content-Length: 2
Origin: http://172.16.35.254
DNT: 1
Connection: keep-alive
Referer: http://172.16.35.254/andi34/

GET /socket.io/?EIO=4&transport=polling&t=NX79GbY&sid=cy3kaa8c_wHcgU-CAAAG HTTP/1.1
Host: 172.16.35.254:14711
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:86.0) Gecko/20100101 Firefox/86.0
Accept: */*
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Origin: http://172.16.35.254
DNT: 1
Connection: keep-alive
Referer: http://172.16.35.254/andi34/

GET /socket.io/?EIO=4&transport=websocket&sid=cy3kaa8c_wHcgU-CAAAG HTTP/1.1
Host: 172.16.35.254:14711
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:86.0) Gecko/20100101 Firefox/86.0
Accept: */*
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Sec-WebSocket-Version: 13
Origin: http://172.16.35.254
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: A/84EIFcApq34FKZzABVJw==
DNT: 1
Connection: keep-alive, Upgrade
Cookie: PHPSESSID=6tq2mligf4869k8r0rj85m4hku
Pragma: no-cache
Cache-Control: no-cache
Upgrade: websocket
```
In parallel I'll get a desktop installed on my Raspi to simulate a local to local connection / setup.

DIY89, 2021-03-19:
Hi.

Now everything is working fine. Many thanks.

I just added the following to my apache2.conf
`<Directory /var/www/>`
`     Order Allow,Deny?`
`     Allow from all`
`     AllowOverride all`
**`     Header always set Access-Control-Allow-Origin "*"`**
**`     Header set Access-Control-Allow-Methods "HEAD, GET, POST, PUT, PATCH, DELETE"`**
**`     Header set Access-Control-Allow-Headers "Origin, Content-Type, X-Auth-Token"`**
`</Directory>`

andi34, 2021-03-21:
Tested on a clean install on Raspberry Pi OS with Desktop using:
```
wget https://gist.githubusercontent.com/andi34/f2c7fe91fb3df92e925e92283e19c073/raw/f80830e6e13fcb1f3512f889eab7b6d3a2edb517/install-raspbian.sh
sudo bash install-raspbian.sh
```
After reboot I've done the setup.

jacques42, 2021-03-21:
Same here, completely reinstalled my raspberry from scratch, using apache. Worked just fine.

closed 2021-03-22 by andi34