andi34/photobooth #204: QR code URL does not consider subfolder installation

jacques42, 2021-03-08:
**Describe the bug**

Generating the QR code, the URL assumes that Photobooth always is installed in the `/var/html/www` folder (document root of the webserver). Installing in a subfolder (i.e. `/var/html/www/photobooth`) QR code URLs won't work 

**To Reproduce**
Steps to reproduce the behavior:
1. Install in a subfolder
2. Go to dev mode, take a picture 
3. Generate QR code
4. Try URL from QR code

**Expected behavior**

The URL encoded in the QR code needs to reflect the actual path of Photobooth installation root. File `api/qrcode.php`

https://github.com/andi34/photobooth/blob/cb07a00e69a97257d55ea109818c39c8b2b0b05b/api/qrcode.php#L12

**Environment (please complete the following information):**
 - OS: iOS, Raspberry Pi
 - Browser : All

**Photobooth**
- Did it work before? no
- Which version of Photobooth are you using? v3 Beta

**Additional context**

andi34, 2021-03-08:
```
$config['webserver']['ip'] = '127.0.0.1/Photobooth';
```

andi34, 2021-03-08:
Or via Webinterface

andi34, 2021-03-08:
Could you try:

```
<?php
require_once '../lib/config.php';

if (!isset($config['webserver']['ip'])) {
    $SERVER_IP = $_SERVER['HTTP_HOST'];
} else {
    $SERVER_IP = $config['webserver']['ip'];
}

$filename = $_GET['filename'];
$api_path = getrootpath('../api');
include '../vendor/phpqrcode/qrlib.php';
$url = 'http://' . $SERVER_IP . $api_path . '/download.php?image=';
QRcode::png($url . $filename, false, QR_ECLEVEL_H, 10);
```

closed 2021-03-10 by andi34