andi34/photobooth #214: Shutter release from gallery/review not working (tested with modern and classic design)

dnai92, 2021-03-23:
Hi,

at first thanks for your nice fork and good work! :)
There´s an issue we can´t get solved. I don´t know if it is a problem ob our setup or a common bug.

**Description:**
We use the DSLR for live preview at the countdown too.
Everything works fine, but if we want to take a photo from the review after a photo is done, it´s not releasing anymore (and there´s no "click" sound from the camera after click the "take photo" button).
There is no preview at the countdown and sometimes no shutter release after cheeeese. Then it shows the common error message.

BUT: If we click on the home/back button on the photo review page, the camera switches back with the "click" sound.
Then from the main page you can take the photo normaly and everything works inclusive preview.

Only taking photos from the button at review is not working correctly.

**Reproduce:**

1. Choose dev Version for install
2. Choose modern or classic design in Admin Panel
3. Take photo from main page
4. Click button "take picture" at review page after shutter release
5. For normal behaviour go back to main page and take photo from there

**Idea:**
I´m not a PHP or a well Java developer, but maybe the code behind the take "photo button" button at the review page does not correctly gives the DSLR free like the button at the main page.
Or ist the preview picture after the shutter release shown directly from cam and not from Raspi?

Thanks in advance :)
Daniel

andi34, 2021-03-23:
Could you please enable dev mode and check your browser console for visible errors?

dnai92, 2021-03-23:
[image]

The DOMException immediately shows when I´ve pressed the button

andi34, 2021-03-23:
With and without "Battery saving mode on gphoto2 live preview" ?

PVTRX, 2021-03-23:
Hey,
dnai92 and I set up the photobooth together, so i would would give a short answer..
Without Battery saving mode.
btw thanks for replying so quickly!

andi34, 2021-03-23:
Could you please test
https://github.com/andi34/photobooth/commit/287ded08a31eba1a1c2b7cd1405cc22db7a94251 and report back?

PVTRX, 2021-03-24:
I copied everything exactly, but unfortunately with no success. The same bug report appears on Chromium.

    // Take Picture Button
    $('.takePic, .newpic').on('click', function (e) {
        e.preventDefault();
        
	if (config.preview.mode == 'gphoto' && !config.preview.gphoto_bsm) {
		api.startVideo('preview');
        }

        api.thrill('photo');
        $('.newpic').blur();
    });

    // Take Collage Button
    $('.takeCollage, .newcollage').on('click', function (e) {
        e.preventDefault();

	if (config.preview.mode == 'gphoto' && !config.preview.gphoto_bsm) {
		api.startVideo('preview');
	}

        api.thrill('collage');
        $('.newcollage').blur();
    });

andi34, 2021-03-24:
After adding you need to run `yarn build` again to get the used transpiled js files. You've done that?

PVTRX, 2021-03-24:
Yes, i did, but nothing really happened. Below you can see the process after i wrote the commend in the console. Or did I do something wrong?

pi@raspberrypi:/var/www/html $ sudo yarn build
yarn run v1.22.5
$ npm-run-all --serial format build:faq build:gulp
$ npm-run-all --parallel format:*
$ prettier src/js/*.js --write
$ prettier {api,lib,test}/{,*/**/}*.php --write
$ prettier src/sass/{,*/**/}*.scss --write
api/admin.php 446ms
api/applyEffects.php 383ms
src/js/admin.js 914ms
api/checkVersion.php 106ms
src/js/adminshortcut.js 111ms
api/chromakeying/save.php 71ms
api/config.php 21ms
src/sass/admin.scss 1093ms
api/deletePhoto.php 191ms
src/sass/classic_chromakeying.scss 152ms
api/deleteTmpPhoto.php 122ms
api/diskusage.php 92ms
api/download.php 72ms
src/sass/classic_live_chromakeying.scss 409ms
api/print.php 280ms
api/qrcode.php 27ms
src/js/chromakeying.js 1026ms
api/rebuildImageDB.php 33ms
api/sendPic.php 131ms
src/sass/classic_style.scss 609ms
api/takePic.php 170ms
src/sass/login.scss 102ms
api/takeVideo.php 83ms
lib/applyFrame.php 28ms
lib/arrayDeepMerge.php 67ms
src/sass/manual.scss 97ms
src/sass/modern_chromakeying.scss 16ms
src/sass/modern_live_chromakeying.scss 57ms
src/sass/modern_style.scss 79ms
src/sass/modules/_theme.scss 90ms
src/sass/partials/_basic.scss 43ms
src/sass/partials/_button.scss 73ms
lib/collage.php 432ms
src/sass/partials/_email.scss 39ms
src/sass/partials/_fonts.scss 71ms
src/sass/partials/_gallery.scss 145ms
lib/config.php 290ms
src/sass/partials/_modal.scss 68ms
src/sass/partials/_modern_button.scss 89ms
src/sass/partials/_ribbon.scss 117ms
src/sass/partials/_scrollbar.scss 32ms
src/js/core.js 1782ms
src/sass/partials/_sidenav.scss 68ms
src/js/diskusage.js 87ms
src/sass/partials/_toggle.scss 100ms
src/js/gallery.js 102ms
src/sass/photoswipe-bottom.scss 173ms
src/js/gallery_updatecheck.js 105ms
lib/configsetup.inc.php 820ms
src/js/i18n-sub.js 26ms
src/js/i18n.js 22ms
lib/db.php 85ms
src/sass/rounded.scss 148ms
src/sass/slideshow.scss 101ms
lib/diskusage.php 73ms
src/sass/vendor/_photoswipe.scss 50ms
src/js/livechroma.js 331ms
lib/filter.php 139ms
src/js/login.js 24ms
lib/hzip.php 47ms
src/js/manual.js 22ms
lib/polaroid.php 62ms
lib/resize.php 95ms
lib/services_start.php 59ms
lib/services_stop.php 38ms
src/js/photoinit.js 262ms
test/collage.php 28ms
src/js/remotebuzzer_server.js 220ms
src/js/slideshow.js 23ms
src/js/sync-to-drive.js 197ms
src/js/theme.js 65ms
$ mdown --input 'faq/*.md' --output manual --header faq/header.html --footer faq/footer.html
$ gulp
[16:19:55] Using gulpfile /var/www/html/gulpfile.js
[16:19:55] Starting 'default'...
[16:19:55] Starting 'sass'...
[16:19:55] Starting 'js'...
[16:20:01] Finished 'sass' after 6.08 s
[16:20:01] Finished 'js' after 6.09 s
[16:20:01] Finished 'default' after 6.1 s
Done in 45.52s.
pi@raspberrypi:/var/www/html $ sudo reboot

andi34, 2021-03-24:
Looks fine to me, you're only missing `sudo chown -R www-data:www-data /var/www/html` but this shouldn't matter.

Does the same happen with gphoto battery saving mode enabled? (Currently can't test gphoto preview myself.)

PVTRX, 2021-03-24:
Thanks for the tip!
Okay with energy saving on it works, but there is the 2000ms delay.
Can the delay decreased somewhere?

andi34, 2021-03-25:
Good to know that it works without bsm. To not have the delay we'll have to create the stream again right after taking the picture. Seems like there's an issue in the logic still, maybe because `api.thrill` does call the `api.reset`? I'll take another look at it and let you know.

andi34, 2021-03-25:
Could you try https://github.com/andi34/photobooth/commit/2340462b84c8f8a6a70d0832b40427cebd390eab without battery saving mode please?

PVTRX, 2021-03-25:
Ok i did and the camera makes the "klick" we want if i push the photo-button in the result window. Unfortunately we have probably adjusted something that we don´t have any preview anymore :(. We need to fix that by ourself because the browser console doesn´t shows any bugrepot. We´ll get back to you this evening.
Thanks for your immediately help!

andi34, 2021-03-25:
You're welcome! You can also visit our Telegram group if you like

https://t.me/PhotoboothGroup

PVTRX, 2021-03-25:
We´ve fixed now the preview and everything works fine now!
Your fix was perfect :)

Btw. after processing the pic, the review gets zoomed in very strong ca. half a second. But after that the pic is shown normally and the buttons appears. Could this be fixed too? Should We open another issue for this?

andi34, 2021-03-25:
Thanks for your reply! I'll merge the fix.

About your other issue: disable "Preload and show image during filter processing"

closed 2021-03-25 by andi34