andi34/photobooth #240: rotary mode AND button mode

h4ns123, 2021-05-22:
The rotary mode is great, but I really miss the possibility to use it simultaneously with the "normal" button mode. Especially when using a normal screen, navigating back is a bit complicated (e.g. from the Gallery) . I think an additional "one-hit-photo-button" would increase the usability value very much.

Is it possible to use different GPIO fort he rotary-encoder and the buttons?

jacques42, 2021-06-25:
@h4ns123 Thanks for the suggestion, I do like the idea. It however took me a while to think it through. Finally I found time to implement a first cut. Branch `v3/feature/parallel-rotary-hwbuttons` / PR https://github.com/andi34/photobooth/pull/262 implements a parallel use of hardware buttons and a rotary encoder. 

Please note this is a config breaking change, when updating. Once you updated, please manually delete your config `config\my.config.inc.php` as well as update you `/boot/config.txt` file. For specific details please see the FAQs (file `faq/faq.md`). You must reboot the Raspi after updating `/boot/config.txt`

Allowing both buttons and rotatry inputs at the same time has a great chance to result in weird behaviors and situations. While I have done quite some testing, I would really appreciate if you can test this functionality in much more detail. That way we can make it most robust, before merge it to DEV. Let me know please if you need help for to install this branch from the git repo.

h4ns123, 2021-06-28:
Hi jacques42! Thank you for adding the functionality so quickly! I will test the solution extensively in the next few days and give you a feedback!

h4ns123, 2021-07-11:
Hi jacques42!
I tested your solution today! However, I had the problem that pressing the buzzer is registered in the log file, but it does not trigger any reaction. The browser log says "remote buzzer client is disabled". Maybe I made a mistake during the installation?

andi34, 2021-07-11:
Followed the steps mentioned here?
https://github.com/andi34/photobooth/blob/v3/feature/parallel-rotary-hwbuttons/faq/faq.md#can-i-use-hardware-button-on-my-raspberry-pi-to-take-a-picture

You've defined the IP Address inside the general section?

jacques42, 2021-07-11:
Did you do a fresh install ? I could imagine it's related to the Config file. Please delete my.config.inc.php, reload the admin panel and reconfigure from scratch. There is a new variable and if it is set to false or
Does not exit the client will throw the error you see.

jacques42, 2021-07-23:
@h4ns123 did you have time to look into this again ?

closed 2021-08-02 by jacques42