andi34/photobooth #254: LED-Buzzer support

h4ns123, 2021-06-06:
Is it possible to provide one or more GPIO for LED in hardware buzzer? It would be pretty cool if the LED goes off during the countdown or flashes increasingly faster and faster.

h4ns123, 2021-07-06:
As further explanation:

I use one of those big Buttons (100 mm) with a built in LED for my photobooth. At the moment I just put the LED on the 3v3 pin so that it is permanently on. The idea was to use a GPIO instead, which is set to HIGH normally and only changes to LOW during the countdown. So you would have a visual feedback when the trigger is working again.

andi34, 2021-08-16:
Hey and sorry for the late response.
You've tried to use the Pre- and Post-command for that?
( https://github.com/andi34/photobooth/pull/232 )
You could for example run a python script with that option. Only make sure permissions are set right.

serpentes80, 2021-09-01:
i wrote 2 python scrips
`import RPi.GPIO as GPIO      # Funktionen fuer die GPIO-Ansteuerung laden
import time                  # fuer sleep() (Pause) noetig

GPIO.setmode(GPIO.BCM)       # die GPIO-Pins im BCM-Modus ansprechen
led=18                       # unsere LED haengt an BCM-Pin 14

GPIO.setup(led, GPIO.OUT)    # in den LED-Pin wird "geschrieben"
GPIO.output(led, GPIO.HIGH)  # LED-Pin auf High (+3.3V) setzen = einschalten
`
this for switching the led on

and `import RPi.GPIO as GPIO      # Funktionen fuer die GPIO-Ansteuerung laden
import time                  # fuer sleep() (Pause) noetig

GPIO.setmode(GPIO.BCM)       # die GPIO-Pins im BCM-Modus ansprechen
led=18                       # unsere LED haengt an BCM-Pin 14

GPIO.setup(led, GPIO.OUT)    # in den LED-Pin wird "geschrieben"
GPIO.output(led, GPIO.LOW)  # LED-Pin auf High (+3.3V) setzen = einschalten
`
for switching the LED off

then I'm running 'python /home/pi/led_pin_18_ein.py' as post and 'python /home/pi/led_pin_18_aus.py' as post photo command

works so far :)

closed 2022-01-09 by stale[bot]