andi34/photobooth #255: Hide all buttons

h4ns123, 2021-06-06:
In case hardware buzzers are used without a touch screen, it would be nice to be able to hide all buttons in the preview as well (I haven't found a way to hide the "filter-button" and the "home-button")

jacques42, 2021-06-10:
Use this switch to turn off filters, the button will be hidden and not visible. 

[image]

Indeed the home button currently can't be hidden. I'll look into it when I find some time

jacques42, 2021-06-10:
PR https://github.com/andi34/photobooth/pull/256 adds a config switch to hide the home button on the results screen.

closed 2021-06-10 by jacques42