andi34/photobooth #258: Ability to change the global font

folkestoneflyer, 2021-06-12:
Currently the font can only be changed for collage/print etc. However I couldn't font the font option in the CSS. 

**Describe the solution you'd like**
Can we have the option to use a custom font for the overall look for the photobooth?

andi34, 2021-06-12:
You can add a own custom style with you own font settings included.

Font is defined here: https://github.com/andi34/photobooth/blob/dev/src/sass/partials/_fonts.scss

Copy all `modern_*.scss` to `custom_*.scss` and make your changes or only partial.

E.g. `custom_style.scss` could look like:
```
@import 'classic_style';

@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 200;
  src: local('Dosis ExtraLight'), local('Dosis-ExtraLight'), url(../fonts/dosis/ExtraLight.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 300;
  src: local('Dosis Light'), local('Dosis-Light'), url(../fonts/dosis/Light.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 400;
  src: local('Dosis Regular'), local('Dosis-Regular'), url(../fonts/dosis/Regular.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 500;
  src: local('Dosis Medium'), local('Dosis-Medium'), url(../fonts/dosis/Medium.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 600;
  src: local('Dosis SemiBold'), local('Dosis-SemiBold'), url(../fonts/dosis/SemiBold.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 700;
  src: local('Dosis Bold'), local('Dosis-Bold'), url(../fonts/dosis/Bold.woff) format('woff');
}
@font-face {
  font-family: 'Dosis';
  font-style: normal;
  font-weight: 800;
  src: local('Dosis ExtraBold'), local('Dosis-ExtraBold'), url(../fonts/dosis/ExtraBold.woff) format('woff');
}
```
(Well, change Dosis to your desired font)

andi34, 2021-06-12:
Once the scss files exist run `yarn build` and choose custom style in admin panel.

andi34, 2021-06-13:
Zum testen:
```
@import 'modern_style';

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 200;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRft6u_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 300;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRf06i_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 400;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpvgxjLBV1hqnzfr-F8sEYMB0Yybp0mudRXdoqqKkkog6SP.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 500;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRfi6m_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 600;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRfp66_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 700;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRfw6-_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

@font-face {
  font-family: 'Barlow Semi Condensed';
  font-style: normal;
  font-weight: 800;
  src: url(https://fonts.gstatic.com/s/barlowsemicondensed/v6/wlpigxjLBV1hqnzfr-F8sEYMB0Yybp0mudRf36y_CWslqZ6GfVK5.woff2) format('woff2');
  unicode-range: U+0100-024F, U+0259, U+1E00-1EFF, U+2020, U+20A0-20AB, U+20AD-20CF, U+2113, U+2C60-2C7F, U+A720-A7FF;
}

body {
  font-family: 'Barlow Semi Condensed', sans-serif;
}
```

closed 2021-07-05 by andi34