andi34/photobooth #259: Hide security-relevant data from frontend

Raukze, 2021-06-21:
**Is your feature request related to a problem? Please describe.**

I thought that the Login feature of this project will prevent others from being able to see my passwords in the config. This is why I was a little concerned when I saw that the entire config file (containing the plain text mail account pwd) is visible to anyone loading any page of Photobooth through the browser. So for me it doesn't make sense to use the mail feature anymore because I know anyone will be able to see my mail password.

**Describe the solution you'd like**

First of all I am not very experienced in web development. What I'd suggest (but do not know if it's technically possible) is to have a separate config file for secrets (such as passwords) which can only be accessed by and used within certain functionalities (e.g. the APIs). So making use of encapsulation is the basic concept I'd suggest to make this do. 

**Describe alternatives you've considered**

**Additional context**

[image]

jacques42, 2021-06-23:
Hey - thanks for your note and I wanted to acknowledge you are spot on. As far as I can see, back when the project started it was designed for functionality, not necessarily with security first in mind. At least that's my read, I may be wrong. For this particular one I agree we should address. 

I'm not necessarily a software security design expert but this then becomes nothing else but a learning opportunity. I appreciate your suggestion and this might be a starting point. Or we could create a server-side secret that is used to encrypt and decrypt on the fly, when the credentials are to be used. Not sure if either ultimately would be a great solution but one or the other way, the credentials should not be accessible by a simple URL, for sure.

andi34, 2021-07-05:
@jacques42 yeah, the project was designed for functionality. 
The Login feature has been added to protect pages from accessing by users, but it has never been a real security feature in view of data protection. 

Sure we should improve security on the mail feature. Every idea and PR is welcome.
Maybe store the mail data in a own config file which is only used by the mail API?

andi34, 2021-07-13:
Please test https://github.com/andi34/photobooth/pull/274

PR #274 (2021-07-19): [security] api: don't show mail password and sensible login data

#### Prerequisites checklist

- [X] I have read the [contributing guidelines](https://github.com/andi34/photobooth/blob/dev/CONTRIBUTING.md).

#### What is the purpose of this pull request? (put an "x" next to an item)

- [X] Security Bug fix; fix #259 

#### What changes did you make? (Give an overview)
We don't need to access the mail password via javascript, same applies for login username and hashed password.
More restriction should be considered.

#### Is there anything you'd like reviewers to focus on?
closed 2021-07-19 by andi34