andi34/photobooth #272: Unexpected return to start page

h4ns123, 2021-07-11:
It happened to me a few times that no image was triggered after the countdown and the view went straight to the start page. I noticed that this happened more often when I reduced the duration of the result page to 10 seconds and started a new countdown after a few seconds.

Is it possible that the command to return to the start page persists even after starting a new countdown and prevents the camera from triggering?

andi34, 2021-07-11:
Please test https://github.com/andi34/photobooth/pull/273

h4ns123, 2021-07-14:
Hi Andi!
The camera always triggers now! Thanks! However, the view still switches to the home page right after and the new photo is not displayed...

andi34, 2021-07-14:
Use a longer time to return to the start page :)

jacques42, 2021-08-02:
Thanks for adding my suggestions into PR#273. I did some extensive testing and where I was able to reprocue the bug, now this PR seems fixes it. 

@h4ns123 can you please also test this PR #273 and see if this fixes the bug for you ?

closed 2021-08-03 by andi34