andi34/photobooth #276: Raspberry Pi as Access Point

Fabian2305, 2021-07-26:
Hey,
first of all thank you very much for sharing the project.
Is it possible to use the Raspberry also as an access point to show the web interface on an Ipad e.g. without using any other networks.

PS: im new in Raspberry, so an instruction how to do it would be great.

Thank you very much

andi34, 2021-07-26:
Hey!
Instructions can be found inside the FAQ of our Wiki:
https://github.com/andi34/photobooth/wiki/FAQ#turn-photobooth-into-a-wifi-hotspot

closed 2021-07-26 by andi34

Fabian2305, 2021-07-27:
Hey, I tryed like the instruction says, but it doesn't work.
I only get "sudo: ./setup-network.sh: command not found"
What did I do wrong?
Thank you

andi34, 2021-07-27:
What version you've installed?
Sure you're trying to run the command inside the right path (/var/www/html/vendor/rpihotspot) ?

Fabian2305, 2021-07-27:
I've installed Option 2, ist Version 3.1.0
Yes I used the right path and checked it a few times, also after reboot

andi34, 2021-07-27:
Does the file exist?
Maybe try
`sudo bash setup-network.sh`

andi34, 2021-07-27:
There's the script, you can also execute it from Desktop for example
https://raw.githubusercontent.com/andi34/rpihotspot/master/setup-network.sh

Fabian2305, 2021-07-27:
I've tryed all your tips, but it still doesn't work and its the same error
message.
First I tested "sudo bash setup-network.sh" and the raspi answered a long
list of code examples. I tryed them all, but without success.
So I used "wget
https://raw.githubusercontent.com/andi34/rpihotspot/master/setup-network.sh"
but it says "Permission denied".
And even after reboot the error "sudo: ./setup-network.sh: command not
found" stays the same!

andi34, 2021-07-27:
https://thepi.io/how-to-use-your-raspberry-pi-as-a-wireless-access-point/

Fabian2305, 2021-07-27:
I've tryed this instruction yesterday, but without any success. Everytime I
tryed to connect a device, my mibilephone or computer says: "no IP-Adress
available" and the connection failed.

andi34, 2021-07-27:
Maybe you start from scratch with a clean raspberry pi is installation?

Fabian2305, 2021-07-27:
I've already tryed this a few times after I tested many different instructions on the internet. I deleted the sd card and installed everything from the beginning.

> Am 27.07.2021 um 23:10 schrieb Andreas Blaesius ***@***.***>:
> 
> Maybe you start from scratch with a clean raspberry pi is installation?

APLtres, 2021-12-21:
Hi, for me worked when I executed this command:

     sudo chmod ug+x ./setup-network.sh

before this comment:

     sudo ./setup-network.sh --install-upgrade --ap-ssid="Photobooth" --ap-password="password" --ap-password-encrypt
--ap-country-code="CA" --ap-ip-address="10.10.10.10" --wifi-interface="wlan0"