andi34/photobooth #284: Enable HID keyboards as hardware buttons

jacques42, 2021-08-02:
**Is your feature request related to a problem? Please describe.**

1) Using the keyboard to trigger a picture / collage, the screen needs to be locally connected to the server system. In scenario's where the screen is i.e. connected through WLAN, to trigger via keyboard does not work. 

2) Many USB devices register as generic HID keyboard to the operating system. 
 
**Describe the solution you'd like**

Enable the use of a HID keyboard devices to trigger pictures / collage. Also support WLAN connected screens. 

**Describe alternatives you've considered**

**Additional context**

andi34, 2021-10-03:
Reopen as my changes don't work if Photobooth isn't opened in a browser.

andi34, 2021-12-27:
Keep open

closed 2022-03-13 by stale[bot]