andi34/photobooth #298: live preview during taking photo from dslr

serpentes80, 2021-08-23:
Hi,

is it just me or is the live preview during taking a photo from dslr not working?
preview mode is set to gphoto2, cam is taking pictures, just no preview :(

andi34, 2021-08-23:
Confirmed working by some German user.
You need to follow the steps mentioned in the FAQ in detail. Also make sure Liveview is supported by GPhoto2 for your camera.

### How to use a live stream as background at countdown?

__(step 1 and 2 in the FAQ, don't matter here__

3. A preview can also be done using the video mode of your DSLR (Linux only), but only works if you access Photobooth via [http://localhost](http://localhost) or [http://127.0.0.1](http://localhost):
    - install all dependencies `sudo apt install ffmpeg v4l2loopback-dkms -y`
    - create a virtual webcam `modprobe v4l2loopback exclusive_caps=1 card_label="GPhoto2 Webcam"`
      - `/dev/video0` is used by default, you can use `v4l2-ctl --list-devices` to check which `/dev/*` is the correct one:  
        If it doesn't match the default setup you need to adjust the `Command to generate a live preview` inside the admin panel!
    - Give permissions to /dev/video* `sudo gpasswd -a www-data video` (this was done automatically if you used the installation script) and reboot once
    - Admin panel config *"Preview mode"*: `from gphoto2`

    **Note**
    - Requires Photobooth v2.11.0 or later!
    - You need to access Photobooth directly via [http://localhost](http://localhost) or [http://127.0.0.1](http://localhost)
    - There's a delay of about 3 seconds until the preview starts, to avoid that disable the `Battery saving mode on gphoto2 live preview` option to generate a preview in background. **This results in a high battery usage and also a general slowdown.**
    - Sometimes Chromium doesn't detect the V4l2 camera launch from php: you need to run `sudo gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0` from terminal first and load Chromium a first time with a webpage asking for the camera.
    - Chromium sometimes has trouble, if there is another webcam like `bcm2835-isp`, it will take it by default instead. Disable other webcams, e.g. `rmmod bcm2835-isp`.
    - To ensure that the configuration works after reboot add the following lines to `/etc/rc.local` (You have to add these lines bevor `exit 0`):
      - `modprobe v4l2loopback exclusive_caps=1 card_label="GPhoto2 Webcam"`
      - `rmmod bcm2835-isp`
    - Make sure the countdown is long enough to start the preview and free gphoto2 at the end of the countdown to be able to take a picture (2 seconds before the countdown ends).
      - For best user experience the countdown should be set at least to 8 seconds.

<hr>

closed 2021-08-23 by andi34

andi34, 2021-08-23:
- Liveview must be supported for your camera model, [check here](http://gphoto.org/proj/libgphoto2/support.php)

serpentes80, 2021-08-24:
worked.
thanks :)

andi34, 2021-08-24:
Glad to hear! If you have further questions you can also join our telegram group 
https://t.me/PhotoboothGroup

(Don't be scared, were talking in German often but English is fine as well :) )