andi34/photobooth #300: Collage Hardware Button

Magicdaniel, 2021-08-31:
Hi, i think i found a bug with the hardware Button for the Collage.

I have two hardware buttons. One for the Picture mode and one for the Collage mode.
If i press the collage button, the first picture will be taken. Because i disabled the automatic mode i have to press again for the second picture, but nothing happens.
At the debug panel this message will pop up: 
buttonActiveCheck error state - requested GPIO 20 , for value 0 but buttonIsPressed: 20

If i can help with screenshots or made something wrong. Tell me pls :)

andi34, 2021-09-27:
From a different conversation:

> I got this error: Remote Buzzer Server [1020]: buttonActiveCheck error state - requested GPIO 21 , for value 0 but buttonIsPressed: 21
> Got it, with an pull-up resistor to 3.3V it works fine. Sorry.
>

jacques42, 2021-12-05:
yes, I can confirm this does not work in the code. Some substantial misses, and needs some work to fix it. I am working this in branch  [bugfix/remotebuzzer-collagebutton](https://github.com/andi34/photobooth/tree/bugfix/remotebuzzer-collagebutton)

jacques42, 2021-12-12:
Will be fixed by PR# https://github.com/andi34/photobooth/pull/351

closed 2021-12-12 by jacques42