andi34/photobooth #309: Picture from the Photobooth are not created

firesta666, 2021-09-08:
Hello all Photobooth users,

please help me. i reinstall the v3 version of the Photobooth and all works fine, only pictures will not displayed after the countdown. The countdown runs fine, the camera take the image but the webpage will not display it. I login to the shell and run the command with the user the webserver is running, the picture will create and rename as the gphoto2 command tell this. Exists any log of the photobooth or log from the raspian? 

I have on friday a birthday party and i would like to bring my Photobooth to the party, but currently it is not working. If anybody can help me intime, contact me and we will find a timeframe to lock together on the system. I have spent money for this project in the past, i love it, my friends love it :-) 
In my situation now i will spent again some Euro's for the project. Please help me, many thanks

[image]

[image]

[image]

[image]

andi34, 2021-09-08:
Have you checked the camera log inside the Debugpanel for issues? I guess it's an permission issue. http://localhost/admin/debugpanel.php

You've updated by hand or did a clean installation (suggested for v3 as a lot of things changed between v3 and older versions)?

I'd suggest running the following commands:
```
sudo chown -R www-data:www-data /var/www/
sudo gpasswd -a www-data plugdev
sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
reboot
```

If still not working:
Maybe you like to join our telegram group? There's a big community who'll help you finding a solution.

andi34, 2021-09-08:
Here's the telegram group: https://t.me/PhotoboothGroup

firesta666, 2021-09-08:
Hello Andy, i will try the debug mode first. Sorry for overseen it :-(.

The debug mode display after take a picture the errors or how i can find them?

I install both versions, manual and the .sh script, same issue with both.

andi34, 2021-09-08:
And maybe you'll tell the issue visible inside the Debugpanel?

closed 2021-09-09 by andi34