andi34/photobooth #310: Resize and positioning from single photos

msmedien, 2021-09-14:
**Is your feature request related to a problem? Please describe.**
I want to use more specific frames for single shot photos. A bit smaller and rotated for a few degrees. Like in the attached example. But i cant find any options for this. Right now i'm only able to use fullsize photos and rotate them.

**Describe the solution you'd like**
The best solution would be some options for the photosize and position in the admin panel. Link shrink: XX%, left-margin: XX%, top-margin: XX% (or in px)

**Describe alternatives you've considered**
As alternativ i could think about some premade templates. (Like the collage option with 2x2, 1+3 and so one.) For example: left-top, left-right, centered, left-bottom, right-bottom.

**Additional context**
Screenshot of a small example.

[image]

der1igedon, 2021-12-29:
This feature would be great! Also for collage configuration.

andi34, 2021-12-29:
Currently I am busy in real life. If someone has time to add this feature I am going to review it. A PR is welcome.

closed 2022-03-13 by stale[bot]