andi34/photobooth #313: yarn build fails on macOS

mhellmeier, 2021-09-19:
**Describe the bug**

The project can't be build on macOS.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repository
2. `yarn install`
3. `yarn build`
4. See error

**Expected behavior**

yarn should also work.

**Environment (please complete the following information):**
 - OS: macOS Big Sur 11.5.2
- yarn version: 1.22.11

**Additional context**

`yarn build` log:
```
yarn run v1.22.11
$ npm-run-all --serial format build:faq build:gulp build:head clean:skipwelcome
$ npm-run-all --parallel format:*
$ prettier {api,lib,test}/{,*/**/}*.php --write
$ prettier src/js/*.js --write
$ prettier src/sass/{,*/**/}*.scss --write
api/admin.php 108ms
api/applyEffects.php 93ms
api/checkConnection.php 14ms
api/checkOS.php 5ms
src/sass/admin.scss 182ms
api/checkVersion.php 10ms
api/config.php 7ms
src/js/admin.js 246ms
src/sass/classic_chromakeying.scss 32ms
api/deletePhoto.php 15ms
api/deleteTmpPhoto.php 10ms
src/js/adminshortcut.js 28ms
api/diskusage.php 17ms
api/download.php 8ms
src/sass/classic_live_chromakeying.scss 59ms
api/print.php 38ms
api/qrcode.php 5ms
api/rebuildImageDB.php 5ms
src/sass/classic_style.scss 73ms
api/sendPic.php 19ms
src/js/chromakeying.js 138ms
api/serverInfo.php 11ms
src/sass/login.scss 19ms
api/shellCommand.php 10ms
src/sass/manual.scss 18ms
src/sass/modern_chromakeying.scss 4ms
src/sass/modern_live_chromakeying.scss 9ms
api/takePic.php 38ms
api/takeVideo.php 10ms
src/sass/modern_style.scss 24ms
api/update.php 9ms
src/sass/photoswipe-bottom.scss 14ms
api/chromakeying/save.php 9ms
src/sass/rounded.scss 18ms
lib/applyFrame.php 5ms
src/sass/slideshow.scss 6ms
lib/applyText.php 4ms
src/sass/update.scss 6ms
src/sass/welcome.scss 6ms
lib/arrayDeepMerge.php 15ms
src/sass/modules/_theme.scss 14ms
src/sass/partials/_basic.scss 10ms
src/sass/partials/_button.scss 13ms
src/sass/partials/_email.scss 8ms
lib/collage.php 75ms
src/sass/partials/_fonts.scss 15ms
src/sass/partials/_gallery.scss 18ms
src/sass/partials/_modal.scss 8ms
lib/config.php 34ms
src/sass/partials/_modern_button.scss 7ms
src/sass/partials/_ribbon.scss 11ms
src/sass/partials/_scrollbar.scss 6ms
src/sass/partials/_sidenav.scss 10ms
src/js/core.js 319ms
src/sass/partials/_toggle.scss 7ms
src/sass/vendor/_photoswipe.scss 8ms
src/js/debugpanel.js 29ms
src/js/dependencies.js 19ms
src/js/diskusage.js 10ms
src/js/gallery.js 5ms
src/js/gallery_updatecheck.js 7ms
src/js/i18n.js 7ms
src/js/livechroma.js 72ms
src/js/login.js 4ms
src/js/manual.js 5ms
lib/configsetup.inc.php 257ms
lib/db.php 12ms
lib/diskusage.php 6ms
lib/filter.php 16ms
lib/hzip.php 10ms
src/js/photoinit.js 69ms
lib/log.php 9ms
lib/polaroid.php 9ms
lib/resize.php 9ms
lib/services_start.php 8ms
src/js/remotebuzzer_client.js 49ms
lib/services_stop.php 7ms
[error] No files matching the pattern were found: "lib/*/**/*.php".
test/collage.php 6ms
[error] No files matching the pattern were found: "test/*/**/*.php".
error Command failed with exit code 2.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
src/js/remotebuzzer_server.js 75ms
src/js/slideshow.js 8ms
src/js/sync-to-drive.jsERROR: "format:php" exited with 2.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
ERROR: "format" exited with 1.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

andi34, 2021-09-19:
No way for me to reproduce, no access to a Mac.
You're able to fix it?

andi34, 2021-09-20:
Could you please try https://github.com/andi34/photobooth/tree/fix/313 and tell if it fixes the issue for you?

closed 2021-09-20 by andi34

andi34, 2021-09-20:
Found a way to let GitHub test on MacOS 11
(https://github.com/andi34/photobooth/commit/1d77dbe652a1864fd52d4921e43b33c41d12bd68)

mhellmeier, 2021-09-21:
I can confirm that it now works on macOS, thanks!

PR #318 (2021-09-20): build: fix build failing on macOS

#### Prerequisites checklist

- [x] I have read the [contributing guidelines](https://github.com/andi34/photobooth/blob/dev/CONTRIBUTING.md).

#### What is the purpose of this pull request? (put an "x" next to an item)

- [x] Bug fix

#### What changes did you make? (Give an overview)
Fix build on MacOS 11.
Fixes #313