andi34/photobooth #317: Wrong language and button size in collage process

mhellmeier, 2021-09-19:
**Describe the bug**

When starting the collage creation process, the Cancel-button has the wrong height. Further, it always shows "Please wait..." above the button (even if the language is changed in the config).

**To Reproduce**
Steps to reproduce the behavior:
1. Change language
2. Adjust font size
3. Make collage

**Expected behavior**

Same button height as the others and a translation of "Please wait".

**Screenshots**

[image]

**Environment (please complete the following information):**
 - OS: Raspberry Pi
 - Browser: Chromium
 - Version: Latest dev branch

**Photobooth**
- Did it work before? no
- Which version of Photobooth are you using? Latest dev branch

mhellmeier, 2021-09-19:
I fixed the button-size problem in the main repo: https://github.com/andreknieriem/photobooth/pull/281

andi34, 2021-09-19:
Thanks, I'll take a look at the button tomorrow. The main repo isn't worked on anymore.

There's no bug regarding the language, "Please wait..." wasn't translated into German yet.
Translation need to be done in crowdin: https://crowdin.com/project/photobooth

andi34, 2021-09-20:
Doesn't seem to fix it and really related to the resolution of your device.

[image]

[image]

My suggestion would be to use a different translation.
Maybe:

weiter,
neu (oder wiederholen?),
abbrechen

What do you think?

andi34, 2021-12-30:
Ich habe die Übersetzung angepasst, bitte aktualisiere auf die aktuelle dev branch gib bescheid ob alles so passt.

closed 2022-01-22 by andi34