andi34/photobooth #319: Possibility to disable "Welcome to your own Photobooth" screen

mhellmeier, 2021-09-25:
**Is your feature request related to a problem? Please describe.**

If I create a new build of the project, the _Welcome to your own Photobooth_ screen always appears at the first time. That's pretty annoying.

**Describe the solution you'd like**

Would be nice to get a possibility to disable this screen (e.g. a config option: "Always disabled welcome screen").

andi34, 2021-09-25:
Must be renamed to `Show welcome screen`, thanks for the finding!
No need for PR. Going to fix it in the evening.

andi34, 2021-09-25:
Personal configuration will always be used on update, if you enable the skip option it won't be visible again until you delete your personal config.

commit edcccea7 (2021-09-25): admin: fix "Skip welcome screen" description

Fix #319
commit d6869b1e (2021-09-25): admin: fix "Skip welcome screen" description

Fix #319

Change-Id: Id0ffc18ded966d739b5b84f3bd4a66a5618719e3
closed 2021-09-25 by andi34