andi34/photobooth #331: Modify Session Timeout or allow specific device to take pictures

F4bsi, 2021-10-04:
**Is your feature request related to a problem? Please describe.**

When using the Login function to protect the main page and only allow logged in devices to take new pictures, the session sometimes will be lost and the device is redirecting to the login page (or the one set in the settings). 

If the device is inside the photobooth and lost its session, nobody will be able to take new pictures (if the front page is protected) until the session is started again (the admin logins).

**Describe the solution you'd like**

I think this is because of the default garbage collection timings from the php.ini which is about 20 to 30 minutes. On Debian garbage collection is done via cron jobs, so even if no one is vising the page the session will be removed. 

I think the session.gc_maxlifetime needs to be increased inside the php.ini file to allow for longer times where nobody is using the device. Only setitng it to a higher value with init_set() probably would not work on debian, because of the default cron solution there.

And also maybe the values for the cookie should be changed too but I'm not 100% sure.

**Describe alternatives you've considered**

Another idea would be to add a function inside the admin panel, to allow setting a specific cookie, that would allow the device having that to access the main page and to take pictures without redirecting the user even if it is not logged in. Probably even better, if the device is inside the photobooth and someone would be able to change the url to the admin page (and because of the active session could do everything). Or we could add another user, that doesnt have admin powers but is able to use the "Take picture" command.

**Additional context**

I think I already used the cookie function in an older version. But it would probably be easier to rewrite it somehow. 
But it should be someting in the line of:
`  $time = time() + (86400 * 30);
   setcookie('take_images', $hash, $time, "/"); // 86400 = 1 day => 30 days`

andi34, 2021-10-04:
Good idea! Currently I am busy with other things on my personal to-do list, would be great if you have time to look at it?

Maybe pickup your old idea for cookie based protection?
https://github.com/F4bsi/photobooth/commits/cookie

Maybe to think about too:
https://www.php.de/forum/webentwicklung/server-hosting-und-workstations/1441903-erledigt-php-session-timeout

andi34, 2021-10-04:
Maybe improve https://github.com/andreknieriem/photobooth/pull/184 ?

andi34, 2021-12-27:
Keep open

closed 2022-03-12 by stale[bot]