andi34/photobooth #365: [BUG]: Remote Buzzer client does not work with HTTPS

jacques42, 2022-01-10:
### Describe the bug

Remote buzzer client is not able to connect to the Photobooth server socket.io process via HTTPS. 

### Steps to reproduce the behavior

Upgrade the web server connection to HTTPS and enable Hardware Buttons. The client is not able to connect

### Expected behavior

Hardware buttons also work when the web server uses HTTPS to communicate

### Photobooth Version

3.x

### Operating System

Raspberry PI OS

### Did it work before?

No

andi34, 2022-03-14:
Keep open

andi34, 2022-06-04:
Still keep open